joedevivo/vscode-circuitpython #1: Installed extension but invoking any command == not found\

codepope, 2020-03-10:


joedevivo, 2020-03-11:
Windows or Linux?

It looks like both are broken because the module I use for serialport communication has native components. I'll need to figure out how to build those for each platform before packaging. I've got some ideas but there are further complications on windows, but I should have something soon that handles those as well.

codepope, 2020-03-11:
macOS. I'll pull the code down and see if I can't get a better handle on it.

joedevivo, 2020-03-11:
Did you upgrade to the latest vscode? The last release did a major version upgrade of electron from 6 to 7 which might have broken some things. If you go to "About Visual Studio Code", it'll give you all the version info

codepope, 2020-03-12:
Yup, it looks like serialport and a module version mismatch.

NightSkySK, 2020-03-12:
Hi I've got the same problem on Windows. 
[image]

On each command from Command Pallet --> Circuit Python: xxx I get 
command == not found
[image]

closed 2020-03-15 by joedevivo

joedevivo, 2020-03-16:
I didn't mean to close this, but 0.0.2 is in and it should work now. Take it for a spin on mac, windows or linux :D

avondohren, 2020-03-17:
Can confirm. `0.0.2` does get things working again on Mac! 🙌 Thanks for your work @joedevivo! I much prefer VS Code to MU Editor, so this extension is amazing!

codepope, 2020-03-17:
Splendid work Joe! Now working. Sorry I couldn't help more in the debug (tho I know way more about the node-gyp issue now).

joedevivo, 2020-03-17:
Me too! If you found anything that looks better than my "fix" of checking in all the bindings I built manually on all these platforms, please let me know. I figured they'll only need to be rebuilt when vscode upgrades electron again, or I upgrade these dependencies.

dzervas, 2020-09-11:
I have this issue in 0.1.2 on Linux. Is there a known workaround?

jwlarocque, 2023-06-10:
In case anyone runs into this in the future, you need to be in an actual workspace which meets the activation requirements.  If you just have a file open and try to run a command it won't work.

PR #3 (2020-03-15): Preparations for 0.0.2
This PR should fix #1 for macOS users and introduce a github action that I'll need to continue working on to support the native bindings on other platforms.