joedevivo/vscode-circuitpython #5: How can I use the REPL?

jonx, 2020-03-20:
Great work so far. I'm using your extension on Win10x64.
Everything seems to be working so far.

Just, how am I supposed to access the REPL?
Maybe the feature is not yet there or maybe I just overlooked it. 

Thanks a bunch,
John

joedevivo, 2020-03-21:
Thanks for the feedback John. The REPL didn't work because of the read-only nature of the output channel; however it was pretty straight forward to convert it to a "Terminal" that can take in input. I just published v0.0.4 which has REPL support. Try it out!

jonx, 2020-03-26:
Great to hear Joe but I was unable to get it to work. That said I'm new to all this so maybe I just missed it:
In the terminal pane I have 2 choices, python and circuit python serial. When I select the latest, all I get is a square:
□
Something must be wrong.
I also don't see the output print by my program anywhere.
Also some commands, those that print something out I guess, don't do anything i.e. CircuitPython: Show Available Libraries. 
I can also still not write in that terminal.
I'm updated to 0.5. Anything I can try?

joedevivo, 2020-04-07:
I just released `0.1.0`. I've seen the terminal act up on windows, but launching vscode as an administrator once fixed it, after that, non-adminstrator could do it also. Something about windows allowing only one thing access to a serial port.

Show available libraries should show a quickpick menu, and picking a library will download it to your workspace's `lib` directory.

askpatrickw, 2020-05-13:
This isn't going to be the _most_ helpful comment. But I've also seen where you connect, nothing streams, CTL-C doesn't do anything but then you restart vscode and it works fine. And then its awesome! ;-)