joedevivo/vscode-circuitpython #9: Add hotkey for opening a library's documentation on readthedocs.io

joedevivo, 2020-03-22:
Every library in the Adafruit bundle has docs up on readthedocs.io. We've already got hover docs, but you may want a more complete picture. Since I haven't figured out how to make a [dash](https://kapeli.com/dash) docset for the Adafruit bundle yet, this will be a good short term solution.

siddacious, 2020-10-31:
There's a pretty specific transform to turn a CP library name into the stub that gets filled in to make the docs URL; basically downcase everything after `Adafruit_CircuitPython_` so 

`Adafruit_CircuitPython_MSA301` => `https://circuitpython.readthedocs.io/projects/msa301/en/latest/`

There are some exceptions but that's a good 80% solution; I'm fairly certain that there is a programatic way of doing this but I'll have to refresh my memory

siddacious, 2020-10-31:
I'd also like to request that Dash not be the primary way of linking to the docs since it appears to be closed source with no free option. 

Of course this is your project so you can do as you wish, however it would be great for everyone to be able to use without being (in essence) behind a paywall