joedevivo/vscode-circuitpython #12: Trying to use on RPi4 on Code - OSS (headmelted)

hoopsurfer, 2020-04-17:
Not able to figure out how to use your extension running on VS Code on my RPi4.  VS Code works great and I can see the CIRCUITPY device and code inside, but any attempt to use the extension generates an error.

joedevivo, 2020-05-09:
Yeah. That's because the plugin is crashing on startup because it can't find native bindings for the serial port library. I've built the native bindings for RPi and they'll be included in `0.1.1` being released today. Let me know if you still have the issue after upgrading. My RPi environment still had some issues, but I also only got the bare minimum up and running to build the bindings.

robertgallup, 2020-09-15:
This is also happening on MacOS (Catalina) for me. All attempts to use the extension result in something like: "command 'circuitpython.openSerialMonitor' not found"

joedevivo, 2020-11-01:
This will need a RPi4 set up at home that Azure can use as builder, since there isn't an RPi builder provided.