joedevivo/vscode-circuitpython #13: VS Code Windows - library display issues...

themdrnsamurai, 2020-04-17:
'circuitpython.library.show' is having issues on a fresh install of the extension. 

[image]

Seeing the same issues with "check for latest bundle".

Thanks for all your work!

joedevivo, 2020-05-09:
I believe this had to do with native bindings for the serial port library, but it should have been resolved in `0.0.2`. I'm releasing `0.1.1` today, let me know if it's still a problem after upgrading

themdrnsamurai, 2020-05-23:
Hey there,

Unfortunately, the behavior is still a problem. "Open serial monitor" also results in the same problem.

GeorgeM987, 2020-09-29:
Yep, have the same issue for pretty much all of the calls...

joedevivo, 2020-09-29:
This should get resolved with #24, and that work is in progress