joedevivo/vscode-circuitpython #14: Local echo in terminal

evansj, 2020-04-22:
Thanks for this plugin. Is there a way to configure local echo in the terminal?

Using VSCode on macOS:

```
>>> 
>>> iimmppoorrtt  ttiimmee

>>> >>> ttiimmee..ttiimmee(())

946687928946687928
```

Double vision. 😄 Cheers

joedevivo, 2020-05-09:
Hrm... that's really odd... I'm not seeing that at all on macOS Catalina. I'm going to release `0.1.1` today. Let me know if you're still seeing this after upgrading.

j9brown, 2022-08-16:
I observed this today while trying out the extension for the first time.  I got a double echo at first, then I got a triple echo (!) after unplugging the board and plugging it back in.  Restarting VSCode seems to have fixed it.