joedevivo/vscode-circuitpython #15: How do I add board a new PID/VID?

jpecor, 2020-05-27:
I'd like to try using this extension for development with our new board.  How is the board metadata file generated?  Is it manual? Or is it automated based on CircuitPython releases?

Our board has been added to CircuitPython and to CP.org.  How can I add it to your extension?

Thanks!

joedevivo, 2020-06-01:
Unfortunately it's still a manual process, but I just pushed 0.1.2, which should have the up to date list. Hopefully I'll be able to leverage azure pipelines going forward, automatically building and releasing whenever a new board is added to adafruit/circuitpython:master, and to update the stubs for the circuit python stdlib whenever a new version is released.

Let me know if 0.1.2 at least covers your board, here's the list from the CHANGELOG:

  - Radomir Dopieralski : Fluff M0
  - Alorium Technology, LLC : AloriumTech Evo M51
  - maholli : PyCubed
  - PJRC : Teensy 4.1
  - Makerdiary : Pitaya Go
  - Nordic Semiconductor : PCA10100
  - HiiBot : HiiBot BlueFi
  - Espressif : Saola 1 w/WROOM
  - Espressif : Saola 1 w/WROVER

jpecor, 2020-06-01:
Hi Joe - yes, this update captured it.  

Thanks!
Jason

closed 2020-06-01 by jpecor

LazaroFilm, 2022-11-17:
Hi, What is the manual procedure for this? I'm using a board that isn't part of the list. Thanks.