joedevivo/vscode-circuitpython #16: Choose CircuitPython Board' Error

jpecor, 2020-06-01:
Hi Joe,

Thanks for adding our board to the VSCode CircuitPython extension.  I tried it out this morning, and I get the following error:
[image]

Any ideas?

Thanks!
Jason

joedevivo, 2020-06-01:
Which OS are you running on?

jpecor, 2020-06-02:
Hi Joe,

macOS 10.14.6

J

jpecor, 2020-06-02:
I am getting an error trying to connect to the serial monitor, as well.

[image]

jpecor, 2020-06-02:
Oh - but I suppose that may be because I can't set the board first?

joedevivo, 2020-06-02:
it seems like they're failing for different reasons.

The first error looks like there's an error happening on the initialization of the extension, which then fails to register the code that is invoked when the selectBoard command is run, but the Open Serial Monitor command is actually running and trying to modify something that doesn't exist. 

 Can you see anything relevant if you open up the extension host log?

[image]

jpecor, 2020-06-02:
In the process of trying to capture some useful output for you, I uninstalled the extension and started from scratch.  After doing so, things seemed to be working.  I could connect to my board, and the serial monitor worked.

But then I closed VSCode, and when I restarted, I got the board connection error.  Uninstalling/reinstalling the CP extension gets it going again.

Here's a screencast of the process if it helps:

https://youtu.be/lpR2FzLd-IU

joedevivo, 2020-06-02:
At this point I think I want to take some time this evening to put together a release with a debug option that can output more useful information. In the mean time, if you can show me the contents of your workspace's .vscode/settings.json in both the working state, and then after you restart.

After that, try doing what you did to reproduce, but add one step. When you're closing vscode, us the trash can icon to manually destroy the serial monitor before exit, it might be holding on to the serial port even after you exit

jpecor, 2020-06-02:
Yes, I can help with that. Probably can't get a look at it until later tonight, though.

joedevivo, 2020-06-02:
Also, give [this](https://stackoverflow.com/questions/44188262/where-are-the-vscode-error-logs-for-extensions) a try and let me know if the console logs are dumping any errors

jpecor, 2020-06-02:
Enabling the Dev Tools and grabbing the error log was quick:

[image]

jpecor, 2020-06-03:
Didn't get to the info capture stuff you requested last night.  I'll do that today, but I need to focus on some other tasks for a bit.  Sorry for the delay.

One thing I realized this morning is that I was not using the extension in a folder/workspace, but just with single files.  Could that be causing a problem?

I'll try it this AM in the context of an open folder and see what I can learn.

joedevivo, 2020-06-03:
Yup! That reproduces is. I'll work on fixing it, but for now open `/Volumes/CIRCUITPY` as a workspace and it's all good.

joedevivo, 2020-06-03:
Oh I'm sorry, I wasn't clear there. It will work opening from ANY workspace folder, but not if you open a `code.py`file on its own. The biggest issue with doing it another way, is that to get the python autocompletion working, I had to write a set of paths in the project's `.vscode/settings.json` file's `python.autoComplete.extraPaths` and I'll need to dig into the python extension to see if there's another way to do it, but I asked about it at the last vscode python twitch stream, and they said it wasn't something that could be set programmatically.

So yeah, while the best experience currently is on the board directly, I do already have some code (that's not currently used) for monitoring physical disks, which should be enough to figure out based on your chosen board (and serial port maybe?) which disk to sync to. 

If you're using a workspace on your computer's disk, all the libraries be downloaded there, and not directly to your device.

The one thing I want to figure out is if it's possible to do a batch operation. As you know, file change on disk will restart the board. If I have to copy ten files, I'd like to copy them all before it restarts, but that might not be possible.

Anyway, give it a try using your whole git project directory as the workspace. That should get you everything except the copying of code.py to the controller. 

^^ that trivial case of "copy only code.py to board" is probably straight forward enough to get in relatively quickly, exposing most of the issues it'll have in the process, like what happens if you have the serial port open when you copy the file...

For the moment, I'd like that to be a manual process. Monitoring the files could wind up copying too often, especially if you've got autosaving set up.

jpecor, 2020-06-03:
Thanks for the clarification.  This helps. 

Yes, I understand wanting to keep the copying as an independent process right now.  Okay by me.

Anything else I can do to help right now?  Thanks for your quick support!

jbroxson, 2021-02-05:
I'm seeing the same thing on Windows 10. 

My output is as follows:

[image]

I am opening the CIRCUITPY volume (E:) directly in VSCode.  None of the paths it is searching for exist in my case.

[image]

Is this any help in resolving it?

Thanks!

joedevivo, 2021-02-06:
@jbroxson Try upgrading the extension to v0.1.8