joedevivo/vscode-circuitpython #18: [BUG] (Question, actually)

jeffdeville, 2020-06-13:
Sorry, not sure where else to ask this. 

It seems like the magic here is dependent on coding directly in the /CIRCUITPYTHON volume. Does everyone just put .git directly into that folder, and if you want to switch projects, just delete and clone the new project in? 

Thanks, this is such a great coding experience all around!

joedevivo, 2020-06-22:
Hi Jeff,

You've got a couple of options. The important thing is that the workspace looks like the `/CIRCUITPYTHON` volume, but it doesn't have to be the volume. What I mean is that as long as it's a directory that can be opened as a VSCode workspace, with a `code.py` in it. If there's a `./lib`, it will update libraries there, and you can copy everything over to the circuit python volume when you want to run it. You can also just open the volume as a workspace, but you're right, I wouldn't want to be managing a .git dir on that Volume.

I personally have a single git repo with my in-flight circuit python projects in sub directories. Then I can copy the whole directory to the volume without worrying about hidden .git files. I'll wind up working on a workspace on the device, and copying it back to my git repo on disk when I'm done working on it.

jeffdeville, 2020-07-06:
Thank you for the input, Joe!

Here's what I've done, in case it's useful to anyone else:

1. Set up a separate directory for my development on my local machine. Make sure I copy the lib/ directory from my device over
2. Install: http://eradman.com/entrproject/ - This is a cross platform file-system watcher
3. Use the following Makefile to keep my device in sync with my local machine. (I'm on a mac)

**NOTE**
I prefer to delete files on the device that are not in my repo. You can do that by adding `--delete ` to the rsync args below. I removed it, out of concern of someone blindly copy/pasting and losing something important that was not a .py or .mpy file.

Makefile
```
sync:
	while sleep 1 ; do find . -name '*.*py' | entr -d rsync -avz --prune-empty-dirs --include="*/" --include="*.*py" --exclude="*" ./ /Volumes/CIRCUITPY ; done
```