joedevivo/vscode-circuitpython #23: [BUG]

mjnorman, 2020-09-11:
Modules built into circuit python show up as unresolved imports.  Just using the basic starting script shows errors on `board` and `digitalio`.

Everything works however.

[image]

askpatrickw, 2020-09-11:
@mjnorman  You might want to update the title of this issue. It just says [BUG]. ;-)

mjnorman, 2020-09-11:
Oops got sidetracked came back and hit submit.  Thanks!

julianrendell, 2020-09-26:
I had the same issue - discovered that selecting any other linter (shift-command/ctrl-p -> python select linter) than pylon makes everything work. Quickly tried flake8, bandit, and prospector.

Seems that pylint doesn't respect the `"python.autoComplete.extraPaths": ` in the workspace config file.

joedevivo, 2020-09-26:
I'll take a look at having it automatically switch off pylint in the extension initialization.

julianrendell, 2020-09-30:
My 2c would be to leave it on, but make a note in the description. Lint'ing is useful, and seems to be required for the autocomplete to work. And for me, any/all the other linters work great with your plugin.

barbudor, 2020-10-28:
Hi
I've been able to (mostly) use pylint with the following tricks :
1) Adding a `.pylintrc` such as : 
```
[MASTER]
init-hook='import sys; sys.path.append("c:/Users/myusername/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.3/boards/0x239A/0x8019");sys.path.append("c:/Users/myusername/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.3/stubs");sys.path.append("c:/Users/myusername/AppData/Roaming/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20201028/adafruit-circuitpython-bundle-py-20201028/lib")'
```
2) changing the stubs file name extensions from `.pyi` files into `.py`

adiazulay, 2021-02-25:
Adafruit has [a good post ](https://learn.adafruit.com/improve-your-code-with-pylint/pylintrc)for how to setup pylint for circuit python. The extension should probably autogenerate this file if it detects pylint is enabled.

adiazulay, 2021-04-09:
@joedevivo, I can take a stab at adding support for pylint