joedevivo/vscode-circuitpython #24: [BUG] Missing `node-v80-darwin-x64/bindings.node` & work-around

julianrendell, 2020-09-26:
OS: MacOS 10.15.6

Added via Visual Studio Code Extension Manager.

Tried to run any CircuitPython commands gave:

```
Activating extension 'joedevivo.vscode-circuitpython' failed: Could not locate the bindings file. Tried:
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/build/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/compiled/12.14.1/darwin/x64/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /Users/julian/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/lib/binding/node-v80-darwin-x64/bindings.node.
_logMessageInConsole @ abstractExtensionService.ts:396
_logOrShowMessage @ abstractExtensionService.ts:396
$onExtensionActivationError @ host.ts:10
_doInvokeHandler @ rpcProtocol.ts:372
_invokeHandler @ rpcProtocol.ts:372
_receiveRequest @ rpcProtocol.ts:253
_receiveOneMessage @ rpcProtocol.ts:185
(anonymous) @ rpcProtocol.ts:51
fire @ event.ts:584
fire @ ipc.net.ts:459
_receiveMessage @ ipc.net.ts:792
(anonymous) @ ipc.net.ts:647
fire @ event.ts:584
acceptChunk @ ipc.net.ts:216
(anonymous) @ ipc.net.ts:137
t @ ipc.net.ts:859
emit @ events.js:223
addChunk @ _stream_readable.js:309
readableAddChunk @ _stream_readable.js:290
Readable.push @ _stream_readable.js:224
onStreamRead @ internal/stream_base_commons.js:181
```

Workaround:

create `.vscode/extensions/joedevivo.vscode-circuitpython-0.1.2/node_modules/@serialport/bindings/lib/binding/node-v80-darwin-x64`
download `https://github.com/serialport/node-serialport/releases/download/%40serialport%2Fbindings%409.0.1/bindings-v9.0.1-electron-v80-darwin-x64.tar.gz`
move the `bindings.node` from the downloaded file to `node-v80-darwin-x64/`

Disable & re-enable. Seems to be getting further.

joedevivo, 2020-09-26:
This is a problem that's plagued me since the beginning. The native bindings for `serialport` change every time vscode upgrades their version of electron. 

Your workaround is what I've been doing during the packaging of the extension, but it won't work with `drivelist`, which is going to be useful for some upcoming features, but they don't publish prebuilds like `serialport`.

I'm hoping to get this all building in azure pipelines soon, which should allow me to prebuild bindings for Windows, macOS, and Linux, and include them in the published build.

The automation will hopefully be able to trigger on new versions of CircuitPython and new boards as well.

pieterv-icloud-com, 2020-09-27:
Getting this now on OS X:

“bindings.node” cannot be opened because the developer cannot be verified.

julianrendell, 2020-09-29:
@pieterv-icloud-com  - I think I ran VS Code from the terminal; not sure if that would avoid triggering that warning though.

julianrendell, 2020-09-29:
@joedevivo Could you add a note somewhere with how to build the extension & what the next steps are to get the CI working are? If there's some small enough/simple enough tasks I maybe able to help.

joedevivo, 2020-09-29:
Thanks for the offer. The reason it's taking so long is the overhead with azure pipelines, but I've finally got it building native bindings for macos, linux, & windows, and packaging them in a .vsix file.

Using that file now won't work because it's not generating the autocomplete stubs, but that should be the last step before I can give you a build to try out. Hopefully I can wrap that up this evening.

joedevivo, 2020-09-30:
@julianrendell I've got it all wired up, try installing [this package](https://artprodeus21.artifacts.visualstudio.com/Ad3f6d785-5e77-4982-8d31-52d4c13c0240/9b023c72-72b6-4ba6-86e5-86bb7df66b40/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2pvZWRldml2by9wcm9qZWN0SWQvOWIwMjNjNzItNzJiNi00YmE2LTg2ZTUtODZiYjdkZjY2YjQwL2J1aWxkSWQvMjgvYXJ0aWZhY3ROYW1lL3BhY2thZ2U1/content?format=zip)

julianrendell, 2020-09-30:
@joedevivo - I uninstalled and removed the one from the extension manager, and installed the package you linked above.

It appears to "just work" - thanks and excellent work! (Noting that I don't have a board with me at the moment... but no binding errors, and I can call and execute functions from the extension.)

I'm not getting the library/autocomplete from stubs I was expecting, but I'm not sure if the stubs are in this build.

But I am a little confused as to how correctly start the extension. I created a python file, but I'm not sure how I enable CP mode. I tried triggering the CP library update and that failed. But starting the "select serial port" command seemed to cause the extension to load, and then commands don't error. 

What is the "correct" way to start a new python file that will have this extension activated?

Thanks again, and looking forward to the full release 😃

julianrendell, 2020-09-30:
Ah- `code.py` is the key I missed. Be aware that I believe there is some thought to changing or adding `code.py` to be something like `runnext.py`. I've seen comments go by on the CircuitPython discord chat, but I'm not on discord enough to keep up with the details.

julianrendell, 2020-09-30:
And `boot.py`?

jensechu, 2020-09-30:
> @julianrendell I've got it all wired up, try installing [this [...]

Thank you so much for doing this. It worked perfectly. 🥇

jensechu, 2020-10-12:
Perhaps you already know but the package linked earlier is also down now. 🙇

joedevivo, 2020-10-13:
@jensechu Give 0.1.3 a try, it should include the fixes from that build.

joedevivo, 2020-10-13:
@julianrendell Could you try out the 0.1.3 release? If that works for you I'll close this.

jensechu, 2020-10-20:
@joedevivo This worked perfectly! Thank you so much. 😄 Please let me know if you create a github sponsor option!

adiazulay, 2021-02-17:
@joedevivo you can use the [node-usb-native](https://www.npmjs.com/package/node-usb-native) package, this what we use in the Arduino extension for vscode.

joedevivo, 2022-03-26:
I recently introduced platform specific builds in 0.1.16. That should resolve this issue permanently.

closed 2022-03-26 by joedevivo