joedevivo/vscode-circuitpython #26: [BUG] stub/board/__init__.pyi and boards/xxx/board.piy conflicts

barbudor, 2020-10-28:
**Describe the bug**
Thanks for your extension, looks very promising 

The stubs for autocompletion and linting are great 👍 👍 

While playing wiht it today, I noticed that you are providing conflicting stubs for `board` module :
- the generic one as `stubs/board/__init__.pyi` which only declares I2C(), SPI() and UART()
- hardware specific `board.pyi` from the `boards/...` tree which declare pins and other hardware specific items

In my case, the completion only takes into account the generic one.
I tried to change the order in `python.autoComplete.extraPaths` without any success

**To Reproduce**
Steps to reproduce the behavior:
1. Use 'CiruitPython: Choose CircuitPython Board' and select your board (in my case CircuitPlayground Express)
2. use this code
```
import board
board.
```
3. The completion list when pressing `.` only list `I2C`, `SPI`, `UART`

[image]

**Expected behavior**
The completion list should show all items from `boards/<VID>/<PID>/board.pyi`

**Workaround**
To get proper completion I followed these steps:
1) rename `stubs/board` as `stubs/board_` to "disable" it
optional steps 2 & 3 to get help on `I2C()`, `SPI()` and `UART()`
2) remove those lines from `boards/.../board.pyi`
```
    I2C: Any = ...
    SPI: Any = ...
    UART: Any = ...
```
3) copy content of `stubs/board/__init__.pyi` at the end of `boards/.../board.pyi`

[image]

[image]

I feel it would be great if you can incorporate this into your process to builds stubs

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version : 20H2
 - VSCode : 1.50.1
 - CircuitPython extension : 0.1.3

**Additional context**

Thanks again for your extension.

joedevivo, 2020-11-04:
@barbudor give [this](https://artprodeus21.artifacts.visualstudio.com/Ad3f6d785-5e77-4982-8d31-52d4c13c0240/9b023c72-72b6-4ba6-86e5-86bb7df66b40/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2pvZWRldml2by9wcm9qZWN0SWQvOWIwMjNjNzItNzJiNi00YmE2LTg2ZTUtODZiYjdkZjY2YjQwL2J1aWxkSWQvNDAvYXJ0aWZhY3ROYW1lL3BhY2thZ2U1/content?format=file&subPath=%2Fvscode-circuitpython-dev-3781beab.vsix) a try

barbudor, 2020-11-05:
Hi. Didn't had time yesterday.
I'll try it no later than this week end

barbudor, 2020-11-07:
Hi @joedevivo 
Looks good now for me
Thanks

PR #27 (2020-11-07): Better stubs
Resolves #26
closed 2020-11-07 by joedevivo