joedevivo/vscode-circuitpython #28: Autocomplete not working

adamstener, 2020-12-04:
**Describe the bug**
I'm able to load this extension properly and run the commands necessary to connect the serial port, select the proper board, etc., but autocomplete never works.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect board
2. Select the proper serial port
3. Select the proper board type
4. Autocomplete does not work

**Expected behavior**
Autocomplete works for circuit-python libraries

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS BigSur
 - Version 11.0.1

ArmoryNode, 2020-12-18:
I recently ran into this issue. It looks like it has to do with the selected Python version. I was able to fix it by changing my Python version to the latest one I had installed.

[image]

adamstener, 2020-12-18:
I've tried updating to the latest and selecting different versions from within VSCode but none of that has worked for me.

adamstener, 2020-12-18:
Command 'CircuitPython: Reload Project Libraries' resulted in an error (The "path" argument must be of type string. Received type object). I see this error when reloading the project libraries, I wonder if this has anything to do with it.

adamstener, 2020-12-23:
I was able to get support working for libraries but not the board yet.  I had to add a setting in VSCode to pick up the directory where this extension stores the stubs.  For some reason though selecting a board doesn't work, VSCode doesn't pick up anything for that at the moment.  If I added another setting to directly look at the right board file, I imagine it would work, and I also imagine that this is probably what the gist of the extension is trying to do, select the right board file and make it accessible in the stub path.  It's just not working for me for some reason.

SAK917, 2021-01-09:
I believe I am also experiencing the same issue on my Windows 10 system.  Am able to open the serial port and select the appropriate board.  However, autocomplete does not work and I am getting the same Command 'CircuitPython: Reload Project Libraries' resulted in an error (The "path" argument must be of type string. Received type object). error adamstener is getting.  I get the same error if I try CircuitPython: List Project Libraries or CircuitPython: Show Available Libraries as well.

jlopez, 2021-01-29:
I had the same `The "path" argument must be of type string. Received type object` error as well. Checking the extension code I found a way to make it work: you just need to make sure a `boot_out.txt` file is present in your project directory. This was my case, since I'm not working directly on the circuitpython volume. Autocomplete now works for me, including `board.` completions.

trepidacious, 2021-02-02:
I had these errors until I restarted vs code - I did also notice some messages about loading bundles, is there any chance this is happening while libraries are downloading?

Secondly, after restarting and being able to see list of libraries etc., I still had imports that couldn't be resolved, however looking in  `.vscode` I noticed that one of the `python.analysis.extraPaths` entries was null:

```json
{
    "python.autoComplete.extraPaths": [
        "/home/rebeam/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.6/boards/0x239A/0x80F4",
        "/home/rebeam/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.6/stubs",
        "/home/rebeam/.config/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20210202/adafruit-circuitpython-bundle-py-20210202/lib"
    ],
    "circuitpython.board.vid": "0x239A",
    "circuitpython.board.pid": "0x80F4",
    "python.analysis.extraPaths": [
        "/home/rebeam/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.6/boards/0x239A/0x80F4",
        "/home/rebeam/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.6/stubs",
        null
    ],
    "circuitpython.board.version": "6.2.0-beta.1"
}
```

Adding the same third path as in `python.autoComplete.extraPaths` resolved this and gave working autocomplete and go-to-definition on the adafruit libraries.

The only remaining import that can't be resolved is `board`, although the boards entries above look correct for the Pico.

wmerkens, 2021-04-23:
I solved the board import by doing a pip3 install Adafruit-Blinka, in fact I installed all of adafruit. adafruit-blinka-displayio , adafruit-circuitpython-framebuf, Adafruit-PlatformDetect, Adafruit-PureIO, adafruit-python-shell.

The real problem here is PyLance & PyLint & Circuitpython extensions are all joined together in sense you need them all working.  The biggest pain is the path's in PyLance, the language engine from MS. It is very path picky and @trepidacious that is exactly what I had in my settings.json file a Null and I was missing the path's to the stubs. Once I fixed those two areas now all the imports worked in the sense that intellisense/intellicode were not complaining or PyLance was not complaining.