joedevivo/vscode-circuitpython #33: Commands not found for CircuitPython

VoltVisionFrenchy, 2021-02-05:
Issue Type: <b>Bug</b>

1) Install the extension
2) Plug in CircuitPython board
3) Run command: "CircuitPython:Open Serial Monitor"

...see screenshot...

Extension version: 0.1.6
VS Code version: Code 1.53.0 (8490d3dde47c57ba65ec40dd192d014fd2113496, 2021-02-03T20:36:38.611Z)
OS version: Windows_NT x64 10.0.19041

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-6500U CPU @ 2.50GHz (4 x 2592)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>protected_video_decode: unavailable_off<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|7.91GB (1.02GB free)|
|Process Argv|--crash-reporter-id 6b1b7e59-e30a-45d8-9c15-f553e93e09b1|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383:30185418
vspor879:30202332
vspor708:30202333
vspor363:30204092
vswsl492:30211401
vstry244:30250195
pythonvsdeb440:30248342
pythonvsded773:30248341
vstes627:30244334
pythonvspyt600cf:30251589

```

</details>

adamstener, 2021-02-05:
does it work if you first select your serial port before opening the monitor?

joedevivo, 2021-02-05:
I suspect the new version of vscode released today uses a new version of electron. I'll update the dependency in the morning

VoltVisionFrenchy, 2021-02-05:
> does it work if you first select your serial port before opening the monitor?

[image]

...after exploring this for a few hours last night I think what is happening is that the extension tries to activate, but fails....therefore none of the commands are available.....I am new to VSCode, so I dont know enough to debug why it is failing yet.

RWhite-EDR, 2021-02-05:
I have your update (v0.1.7), but still nothing...
```
Version: 1.53.0
Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
Date: 2021-02-03T16:01:04.046Z (1 day ago)
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 20.3.0
```

VoltVisionFrenchy, 2021-02-05:
> I have your update (v0.1.7), but still nothing...
> 
> ```
> Version: 1.53.0
> Commit: 8490d3dde47c57ba65ec40dd192d014fd2113496
> Date: 2021-02-03T16:01:04.046Z (1 day ago)
> Electron: 11.2.1
> Chrome: 87.0.4280.141
> Node.js: 12.18.3
> V8: 8.7.220.31-electron.0
> OS: Darwin x64 20.3.0
> ```

@RWhite-EDR do you have the same problem?  I agree that the new v0.1.7 update did not fix my issue.

VoltVisionFrenchy, 2021-02-05:
Sorry if I am wasting anyone's time with newbie ignorance here, but I am starting to wonder if my problem is related to these?
[image]

PID seems to be missing?
[image]

[image]

[image]

I will try manually updating these and see what happens.

adamstener, 2021-02-05:
Ideally these values would be set when you choose your board.  Having to manually change these settings every time you plug in a new board and select it would be cumbersome.  I think that was the intention of that command.

VoltVisionFrenchy, 2021-02-05:
> Ideally these values would be set when you choose your board. Having to manually change these settings every time you plug in a new board and select it would be cumbersome. I think that was the intention of that command.

So its a catch-22:
[image]

adamstener, 2021-02-05:
It's not a catch 22 I think it's just not working correctly.  You can manually set these values in settings but I don't think that is ideal nor do I think that will magically make this work.

VoltVisionFrenchy, 2021-02-05:
> It's not a catch 22 I think it's just not working correctly. You can manually set these values in settings but I don't think that is ideal nor do I think that will magically make this work.

@adamstener is yours broken too or am I the only one? (not sure about that yet).

adamstener, 2021-02-05:
Mine is broken but in a slightly different way.  I was able to get past the choose board and choose port errors, though I'm not up to date with the latest versions. However, even though these work, there are still other issues I'm having with fetching libraries, etc, and while the select board command doesn't error, it doesn't exactly work either.  I _believe_ the intention is to select a board which will copy the right support file over so that you can have board. completions with the correct pins etc.  Despite my command not erroring, I don't get completion for board values and it doesn't work today.  I am however able to select the proper port and see the serial monitor working.

RWhite-EDR, 2021-02-05:
@VoltVisionFrenchy @adamstener Mine is also broken.  I'm not able to selectBoard, selectSerialPort, and openSerialMonitor are all 'not found'.

I've tried removing and reinstalling the extension, starting and stopping VS Code, etc.

Still poking around.

Workaround:  I normally use deploy script to move code to the board and I'm using the mac screen command to view the serial monitor.

VoltVisionFrenchy, 2021-02-05:
> I've tried removing and reinstalling the extension, starting and stopping VS Code, etc.

Newbie question:
Does VSCode have a debug window that shows what the error is and why it cannot load the extension?  I think the commands are failing because the extension isnt loading properly....but VSCode doesnt say why the extension isnt loading...

adamstener, 2021-02-05:
I accidentally quit VSCode on one of my laptops and it automatically updated.  I'm now in the same boat here, all commands fail.

joedevivo, 2021-02-05:
0.1.8 is running now, and should resolve the issue. I had forgotten to add the new version of electron to the script that actually builds the bindings.

VoltVisionFrenchy, 2021-02-06:
yes, confirmed....v0.1.8 seems good!!!!!   Select serial port worked and recognized the board on the first time.....Open serial port worked first time. 
 Thanks!!!!!

closed 2021-02-06 by VoltVisionFrenchy

theLMGN, 2021-03-02:
It's still broken for me :/

```
Version: 1.53.2
Commit: 622cb03f7e070a9670c94bae1a45d78d7181fbd4
Date: 2021-02-11T11:45:54.515Z (2 wks ago)
Electron: 11.2.1
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 19.6.0
```