joedevivo/vscode-circuitpython #34: [BUG] Error installing library using "Show Available Libraries" command.

BeldrothTheGold, 2021-02-06:
**Describe the bug**
When attempting to add a library to my project, I get an error. 

Command 'CircuitPython: Show Available Libraries' resulted in an error (Running the contributed command: 'circuitpython.library.show' failed.)

Output from the extension host is:
[2021-02-06 00:17:32.475] [exthost] [error] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at validateString (internal/validators.js:120:11)
	at Object.join (path.js:1039:7)
	at LibraryQP.install (/Users/scott/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:64:41)
	at LibraryQP.onClick (/Users/scott/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:51:22)
	at LibraryManager.<anonymous> (/Users/scott/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:135:24)
	at Generator.next (<anonymous>)
	at fulfilled (/Users/scott/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:5:58)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:97:5) circuitpython

**To Reproduce**
Steps to reproduce the behavior:
1. Open project
2. Use command pallet to run Show Available Libraries command
3. Select library
4. error

**Expected behavior**
I believe that selecting an available library is supposed to copy it to the circuitpy drive?

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Big Sur
 - Version 11.2

G7OCD, 2021-02-17:
Also seeing this. Code v1.53.2,  CircuitPython extn v0.1.8,  W10 build 19042

geekGeeBM, 2021-03-27:
I'm having the same issue on Xubuntu 20.04 using VSCode 1.54.3 with Extension v0.1.8.

thousandmoths, 2021-03-27:
I'm getting errors on all callables also, libraries bundle updates at start but can't list, reload  or anything else,.

`[2021-03-27 22:22:35.798] [exthost] [info] ExtensionService#loadCommonJSModule file:///c:/Users/gary/.vscode/extensions/ms-vscode-remote.remote-wsl-0.54.6/dist/extension
[2021-03-27 22:22:54.956] [exthost] [error] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at validateString (internal/validators.js:120:11)
	at Object.resolve (path.js:139:9)
	at ProviderAsync._getRootDirectory (c:\Users\gary\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.8\node_modules\fast-glob\out\providers\provider.js:17:21)
	at ProviderAsync.read (c:\Users\gary\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.8\node_modules\fast-glob\out\providers\async.js:11:27)
	at Array.map (<anonymous>)
	at getWorks (c:\Users\gary\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.8\node_modules\fast-glob\out\index.js:58:18)
	at FastGlob (c:\Users\gary\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.8\node_modules\fast-glob\out\index.js:10:19)
	at c:\Users\gary\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.8\node_modules\globby\index.js:133:52
	at Array.map (<anonymous>)
	at module.exports (c:\Users\gary\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.8\node_modules\globby\index.js:133:40) circuitpython.library.reload`

thousandmoths, 2021-03-29:
Having a look into the errors generated, is this basically because the extension needs an update for VSC v 1.54.3?

closed 2021-03-31 by joedevivo

patridge, 2021-09-09:
Was there a workaround somewhere I missed? I'm seeing this same issue on Windows 10 with extension v0.1.15.

Karolis92, 2022-02-14:
Getting same error.  How to fix it?

jango-fx, 2022-03-21:
same here on macOS Monterey 12.2.1 (21D62), VSC Version: 1.65.2, vscode-circuitpython v0.1.15 :(

jango-fx, 2022-03-21:
ok, the error message says

`Command 'CircuitPython: Show Available Libraries' resulted in an error (A system error occurred (ENOENT: no such file or directory, stat '/Users/felixg/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20220321/adafruit-circuitpython-bundle-7.x-mpy-20220321/lib/adafruit_lsm6ds'))`

the actual lib folder is at
`~/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20220321/adafruit-circuitpython-bundle-py-20220321/lib/adafruit_lsm6ds`
though.

jango-fx, 2022-03-21:
super dirty quickfix: rename the actual folder to what the installer expects 🤷