joedevivo/vscode-circuitpython #35: [BUG] Unable to list, install, or update libraries

tylercrumpton, 2021-03-08:
**Describe the bug**
The extension currently throws an error any time I try to list, update, or install libraries. An error message pops up during any of these actions with a message like:

```
Command 'CircuitPython: Show Available Libraries' resulted in an error (Running the contributed command: 'circuitpython.library.show' failed.)
```

```
Command 'CircuitPython: Update All Libraries' resulted in an error (Running the contributed command: 'circuitpython.library.update' failed.)
```

```
Command 'CircuitPython: List Project Libraries' resulted in an error (Running the contributed command: 'circuitpython.library.list' failed.)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Open Command Palette
2. Select any of CircuitPython > Show Available Libraries, List Project Libraries, or Update All Libraries

**Screenshots**
[image]

**Logs**
```
[2021-03-08 10:36:54.339] [exthost] [error] TypeError: Cannot read property 'name' of undefined
	at new LibraryQP (/Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:58:20)
	at /Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:206:22
	at Array.forEach (<anonymous>)
	at LibraryManager.getInstalledChoices (/Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:203:8)
	at LibraryManager.getAllChoices (/Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:195:33)
	at LibraryManager.<anonymous> (/Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:170:26)
	at Generator.next (<anonymous>)
	at /Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:32:69
	at new Promise (<anonymous>)
	at __awaiter (/Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:12:12)
	at LibraryManager.show (/Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/librarymanager/libraryManager.js:169:12)
	at /Users/tyler.crumpton/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.8/out/container.js:51:77
	at b._executeContributedCommand (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:86:88517)
	at b.$executeContributedCommand (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:86:88849)
	at p._doInvokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:10276)
	at p._invokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:9968)
	at p._receiveRequest (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:8638)
	at p._receiveOneMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:7440)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:5568
	at v.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)
	at d.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:15515)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:29104
	at v.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)
	at d.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:15515)
	at t._receiveMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:20765)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:17659
	at v.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:55:1836)
	at u.acceptChunk (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:12880)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:63:12228
	at Socket.m (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:104:12375)
	at Socket.emit (events.js:315:20)
	at Socket.EventEmitter.emit (domain.js:483:12)
	at addChunk (_stream_readable.js:295:12)
	at readableAddChunk (_stream_readable.js:271:9)
	at Socket.Readable.push (_stream_readable.js:212:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:186:23) circuitpython.library.show
```

**Desktop (please complete the following information):**
 - OS: macOS Catalina Version 10.15.7
 - VSCode version: 
```
Commit: f30a9b73e8ffc278e71575118b6bf568f04587c8
Date: 2021-03-04T22:42:18.719Z (3 days ago)
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin x64 19.6.0
```

**Additional context**
CircuitPython Extension version: v0.1.8

It looks like it might be similar to the #34 that is open already, but it's resulting in a different error.

makermelissa, 2021-03-30:
I've been looking into this. Here's what I know so far. It seems to get hung up inside of the loadLibraryMetadata() function of the libraryManager.ts file. I think it's having trouble parsing the library for some reason, so there is no data to list or update with.

makermelissa, 2021-03-30:
Ok, I figured out the culprit. The https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/ library is structured differently than other libraries in that it is 2 folders deep and has no version information at 1 layer deep which throws the plugin off. It was first published 27 days ago, which fits in with the timeline of the issue.

closed 2021-03-31 by joedevivo