joedevivo/vscode-circuitpython #36: Command 'CircuitPython: List Project Libraries' resulted in an error (command 'circuitpython.library.list' not found)[BUG]

chuckuntulis, 2021-03-28:
I have installed CircuitPython v0.1.8 extension in VSCode on two Windows 10 systems and they both report the error. I am running VSCode 1.54.3 on both systems. I am trying to develop code for Raspberry Pi Pico. I have used Mu to devlop code but I much prefer VSCode.

I do a View>Command Palette>CircuitPython: List Project Libraries and "Command 'CircuitPython: List Project Libraries' resulted in an error (command 'circuitpython.library.list' not found)[BUG]". Similar errors are reported for other CircuitPython commands?

closed 2021-03-31 by joedevivo

chuckuntulis, 2021-03-31:
Hi Joe,
I am still having the problem on one of my machines. I agree that it is
probably not related to your code. Do you have any idea why vscode would
report an error or where I might look to see what the problem is on my
system? Thanks

joedevivo, 2021-03-31:
I'd start by downloading the version of the extension built from the PR [here](https://dev.azure.com/joedevivo/vscode-circuitpython/_build/results?buildId=56&view=artifacts&pathAsName=false&type=publishedArtifacts).

I haven't published a version with this fix in it yet, because I identified an edge case where the extension still doesn't work when you've never downloaded the bundle before. Hopefully I'll be able to figure that out this afternoon.

chuckuntulis, 2021-03-31:
Thanks. I will try it later this afternoon.

As I have been trying different options, I consistently  see "Python
extension loading" with a rotating arrow in front of it. This never stops
unless I exit VSCode. I have no idea why it does not exit or error out. I
have checked the Python paths and can execute Python and py in the Terminal
in VSCode and in Windows PowerShell. Is there some other path that might
not be set correctly? I do not know where to look for a VSCode support
group. Where do you go/look for VSCode issues?

I tried planting some error prints in the Activator.js for Pico-go, but
when executing a view>Command Palette> pico-go command the error message
appears before getting to that point. Do you specify a Python extension
dependency in your code and where is it specified?

Again, your code and Pico-go execute properly on my other system so there
must be something upstream that is different on the two systems. I do not
know what the right question is to ask or whom to ask it of.