joedevivo/vscode-circuitpython #38: Extension not working on M1 Big Sur

wildestpixel, 2021-04-01:
Version 0.19 of extension, Universal app 1.55

any attempt to load any of the extension parts results in errors such as 

`Command 'CircuitPython: Select Serial Port' resulted in an error (command 'circuitpython.selectSerialPort' not found)`

joedevivo, 2021-04-01:
Can open the `Output` pane at the bottom and select `Log (Extension Host)`, and then look for this error message?

```
[2021-04-01 07:50:42.433] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-04-01 07:50:42.433] [exthost] [error] Error: dlopen(/Users/joe/Code/vscode-circuitpython/node_modules/@serialport/bindings/build/Release/bindings.node, 1): no suitable image found.  Did find:
	/Users/joe/Code/vscode-circuitpython/node_modules/@serialport/bindings/build/Release/bindings.node: mach-o, but wrong architecture
```

[image]

I think I've been running the Intel build of VSCode on my M1, which worked fine. Now that I've downloaded the Universal build, I'm seeing this error. If that's what you're seeing as well, then I've got something to go on.

wildestpixel, 2021-04-01:
it is indeed somewhat different - 

`[2021-04-01 18:04:36.513] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-04-01 18:04:36.513] [exthost] [error] Error: Could not locate the bindings file. Tried:
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/build/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/build/Release/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/Debug/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/out/Release/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/Release/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/build/default/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/compiled/12.18.3/darwin/arm64/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64/bindings.node
	at bindings (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/bindings/bindings.js:126:9)
	at Object.<anonymous> (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/darwin.js:2:36)
	at Module.u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1248)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1220:10)
	at Module.load (internal/modules/cjs/loader.js:1039:32)
	at Module._load (internal/modules/cjs/loader.js:932:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12738)
	at Function.o._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:28124)
	at Function.E._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:24842)
	at Function.a._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:95:59194)
	at Module.require (internal/modules/cjs/loader.js:1079:19)
	at v (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:694)
	at Object.<anonymous> (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/index.js:10:22)
	at Module.u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1248)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1220:10)
	at Module.load (internal/modules/cjs/loader.js:1039:32)
	at Module._load (internal/modules/cjs/loader.js:932:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12738)
	at Function.o._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:28124)
	at Function.E._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:24842)
	at Function.a._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:95:59194)
	at Module.require (internal/modules/cjs/loader.js:1079:19)
	at v (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:694)
	at Object.<anonymous> (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/serialport/lib/index.js:2:17)
	at Module.u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1248)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1220:10)
	at Module.load (internal/modules/cjs/loader.js:1039:32)
	at Module._load (internal/modules/cjs/loader.js:932:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12738)
	at Function.o._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:28124)
	at Function.E._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:24842)
	at Function.a._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:95:59194)
	at Module.require (internal/modules/cjs/loader.js:1079:19)
[... cut ...]
	at h._loadCommonJSModule (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:26251)
	at h._doActivateExtension (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:91:12016)
	at h._activateExtension (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:91:11072)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
	at async Promise.all (index 0)`

joedevivo, 2021-04-01:
That's great actually. I think the root cause is the same, in that both of us are looking for the arm64 bindings.node file, yours can't find any bindings.node file, and mine is finding one, but it's an intel build artifact. Making sure we both have access to a macOS arm64 bindings.node will fix it for both of us, but figuring out how to get it... that's something else.

Not to go into too much detail, but I have an azure pipeline job for each linux, windows, and mac (all on intel) to build the bindings for each, and then package them all up in the end. Those are the only runtimes I have access to there, which is why I haven't supported running on RPi either. I've got a couple of things to try out here, but it may be a few days. I've got a hack solution I can put in over the weekend, but I'd like to try and figure out the right way to do this.

wildestpixel, 2021-04-01:
Thats great news Joe - I'm late to the party with VSCode having predominantly used other "python specific editors", and only at the point did I switch to VSCode did I also switch to an Apple Silicon Mac and hence came up against this, but didn't git issue it until now

joedevivo, 2021-04-02:
Try installing [this build](https://artprodeus21.artifacts.visualstudio.com/Ad3f6d785-5e77-4982-8d31-52d4c13c0240/9b023c72-72b6-4ba6-86e5-86bb7df66b40/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2pvZWRldml2by9wcm9qZWN0SWQvOWIwMjNjNzItNzJiNi00YmE2LTg2ZTUtODZiYjdkZjY2YjQwL2J1aWxkSWQvNjIvYXJ0aWZhY3ROYW1lL3BhY2thZ2U1/content?format=file&subPath=%2Fvscode-circuitpython-dev-96d50d2c.vsix) manually. If it resolves this for you I'll figure out how to make it a permanent part of the build process.

wildestpixel, 2021-04-02:
Now my problem from the log seems to mirror that of yours 

`[2021-04-02 16:23:33.841] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-04-02 16:23:33.841] [exthost] [error] Error: dlopen(/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64/bindings.node, 1): no suitable image found.  Did find:
	/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64/bindings.node: mach-o, but wrong architecture
	/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64/bindings.node: mach-o, but wrong architecture`

from your switch to the universal app. I can see the node file within a tarball, but don't know whether they are symlinked correctly or not active as they are still tarballed - unfortunately very much above my head!

joedevivo, 2021-04-03:
Those tarballs are actually leftover from the build process. If you look in this path: `/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding` you'll see all of the versions for the different platforms. So, your error message is at least showing that it's looking in the right place.

I saw that error once, but then after restarting vscode, it went away. Looking back, what I think happened is that I had the Apple Silicon build installed, and when vscode updated, it installed the universal build. Do you know which version (Silicon vs Universal) you have installed?

If it's not Universal, the fastest way for you to fix this would be to install the Universal app for the time being. You don't even need to replace what you have in `/Applications`, you can run the app straight from `~/Downloads` just to see if it works.

I don't really have much more to report. It took me longer than I'd have liked to realize that the update switched me to Universal.

wildestpixel, 2021-04-04:
Been on universal build from the start on this Mac

meglorin, 2021-04-04:
Just wanted to add another user to test potential solutions for this issue.  I get the same results as @wildestpixel when I installed the build you put up:

```
[2021-04-04 13:58:12.090] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-04-04 13:58:12.091] [exthost] [error] Error: dlopen(/Users/tsp/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64/bindings.node, 1): no suitable image found.  Did find:
	/Users/tsp/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64/bindings.node: mach-o, but wrong architecture
	/Users/tsp/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/node_modules/@serialport/bindings/lib/binding/node-v85-darwin-arm64/bindings.node: mach-o, but wrong architecture
```

Version: 1.55.0
Commit: c185983a683d14c396952dd432459097bc7f757f
Date: 2021-03-30T16:07:58.402Z (5 days ago)
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Darwin arm64 20.3.0

joedevivo, 2021-04-04:
Have either of you installed Rosetta2?

wildestpixel, 2021-04-05:
Yes I’m usIng it to run quite a few intel apps

meglorin, 2021-04-05:
Like @wildestpixel, I had to install it for apps that haven't made the move over yet.

joedevivo, 2021-04-06:
Think I got it. Try [this build](https://artprodeus21.artifacts.visualstudio.com/Ad3f6d785-5e77-4982-8d31-52d4c13c0240/9b023c72-72b6-4ba6-86e5-86bb7df66b40/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2pvZWRldml2by9wcm9qZWN0SWQvOWIwMjNjNzItNzJiNi00YmE2LTg2ZTUtODZiYjdkZjY2YjQwL2J1aWxkSWQvNzQvYXJ0aWZhY3ROYW1lL3BhY2thZ2U1/content?format=file&subPath=%2Fvscode-circuitpython-dev-1c9f3531.vsix)

wildestpixel, 2021-04-06:
Instant success - installed over the top of previous extension didn’t need to flush the destination. Many thanks :)

closed 2021-04-06 by joedevivo

meglorin, 2021-04-07:
Confirming that it worked here.  I uninstalled first and then installed the latest build.  It is unhappy if you try to do this at work and don't actually have a device plugged in when you attempt to open the serial port... but that is my issue, not yours.

PR #43 (2021-04-06): Try local m1 builder
Build bindings on my mac mini until azure gets arm64 macOS vms.

Fixes #38