joedevivo/vscode-circuitpython #42: [BUG]

Idirianuk, 2021-04-02:
Error at startup

Steps to reproduce the behavior:
Start vs.code with extension enabled

Expected behavior:
Extension to load and initialise successfully
Windows 10

Log output:

`
[2021-04-02 09:48:50.950] [exthost] [error] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at validateString (internal/validators.js:120:11)
	at Object.join (path.js:375:7)
	at LibraryManager.getProjectLibDir (c:\Users\John\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:229:22)
	at LibraryManager.<anonymous> (c:\Users\John\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:114:39)
	at Generator.next (<anonymous>)
	at fulfilled (c:\Users\John\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:5:58)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
`

Idirianuk, 2021-04-02:
Tried the extension on the actual circuitpython drive (rpi Pico) and it all works perfectly. I'd been working from a hard drive directory then using a filesync program to copy to the Pico. Doing this still gives the same error indicated above. Might this be something to do with the fact that insted of working a a directory level of "P:\" (for example) I'm working at a lave of "D:\Code\adafruit_project\"?

joedevivo, 2021-04-07:
Try v0.1.11. I just released it to fix #44 and I think that might be what's happening here as well.

Idirianuk, 2021-04-07:
> 
> 
> Try v0.1.11. I just released it to fix #44 and I think that might be what's happening here as well.

I think this is now fixed. Thank you for your brilliant efforts! :)

closed 2021-04-07 by Idirianuk