joedevivo/vscode-circuitpython #44: Extension fails to load on Mac OS Catalina

creativeprojects, 2021-04-05:
**Describe the bug**
The extension apparently fails to load on Mac OS Catalina (10.15.7). When trying to select a board, I get the error `command 'circuitpython.selectBoard' not found`

**To Reproduce**
- Start vscode in an empty folder.
- Copy a working `code.py` (default one copied from a brand new FeatherS2).
- Install the extension
- Try to select a board

**Expected behavior**
Commands to select a board should have been installed by the extension

**Desktop (please complete the following information):**
 - Mac OS Catalina - Mac mini 2018
 - Version 10.15.7

**Additional context**
From the extension host logs, I can see:
```
[2021-04-05 12:22:41.194] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-04-05 12:22:41.194] [exthost] [error] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at validateString (internal/validators.js:120:11)
	at Object.join (path.js:1039:7)
	at LibraryManager.getProjectLibDir (/Users/cp/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/out/librarymanager/libraryManager.js:229:22)
	at LibraryManager.<anonymous> (/Users/cp/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/out/librarymanager/libraryManager.js:114:39)
	at Generator.next (<anonymous>)
	at fulfilled (/Users/cp/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/out/librarymanager/libraryManager.js:5:58)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
```

and a `settings.json` file has been generated:

```json
{
    "python.languageServer": "Pylance",
    "python.linting.pylintEnabled": false,
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none"
    },
    "python.analysis.extraPaths": [
        null,
        "/Users/cp/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.10/stubs",
        "/Users/cp/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20210405/adafruit-circuitpython-bundle-py-20210405/lib"
    ]
}
```

I thought that the `null` path was a bit funny, so I deleted the line. Then I reload vscode window, the extension fails, and the `null` path is added back in the `settings.json` file

I also tried with de-activating the Arduino extension that I'm using for another board, in case there was some kind of incompatibility, but it didn't change anything sadly.

After writing all of that, I just realise the Pylance is a new language server for vscode and maybe it's not supported?

Thanks for your help :)

creativeprojects, 2021-04-05:
I think I found the issue!

I was reading the code in `libraryManager.js` and I realised the file `boot_out.txt` is needed otherwise it won't detect the project folder properly.

So now it works with a `boot_out.txt` in my workspace :)

I suggest you could throw an error message saying the file was not found or something 👍🏻

joedevivo, 2021-04-07:
Thanks for the detective work. I think this is fixed in v0.1.11. Let me know 😀

creativeprojects, 2021-04-12:
I can confirm it works without a `boot_out.txt`. Thanks for fixing this quickly 👍🏻

closed 2021-04-12 by creativeprojects