joedevivo/vscode-circuitpython #45: Error running library related commands

SAK917, 2021-04-06:
**Describe the bug**
When trying to run Show Available Libraries, List Project Libraries and Update All Libraries, I get the following error:

[image]

**To Reproduce**
1. In VS Code click on File > Open Folder... and select the CIRCUITPY (D:) drive
2. Open the code.py file
3. View > Command Palette... > CircuitPython: List Project Libraries (or Update All Libraries, or Show Available Libraries)

**Desktop (please complete the following information):**
CircuitPython Extension:  v0.1.10

OS: Windows 10 build 19042.870

VS Code: 
Version: 1.55.0 (user setup)
Commit: c185983a683d14c396952dd432459097bc7f757f
Date: 2021-03-30T16:01:55.261Z
Electron: 11.3.0
Chrome: 87.0.4280.141
Node.js: 12.18.3
V8: 8.7.220.31-electron.0
OS: Windows_NT x64 10.0.19042

**Additional context**
```
[2021-04-06 15:35:29.256] [exthost] [error] TypeError: Cannot read property 'name' of undefined
	at new LibraryQP (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:41:24)
	at c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:172:26
	at Array.forEach (<anonymous>)
	at LibraryManager.getInstalledChoices (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:169:59)
	at LibraryManager.<anonymous> (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:147:32)
	at Generator.next (<anonymous>)
	at c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:8:71
	at new Promise (<anonymous>)
	at __awaiter (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:4:12)
	at LibraryManager.list (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\librarymanager\libraryManager.js:146:16)
	at c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.10\out\container.js:52:77
	at S._executeContributedCommand (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:86:106770)
	at S.$executeContributedCommand (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:86:107208)
	at u._doInvokeHandler (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:12794)
	at u._invokeHandler (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:12478)
	at u._receiveRequest (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:11145)
	at u._receiveOneMessage (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:9931)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:8032
	at _.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:57:1836)
	at E.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:15496)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:105:30083
	at _.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:57:1836)
	at E.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:15496)
	at t._receiveMessage (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:20754)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:17640
	at _.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:57:1836)
	at acceptChunk (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:12861)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:12209
	at Socket.w (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:105:13173)
	at Socket.emit (events.js:315:20)
	at addChunk (_stream_readable.js:295:12)
	at readableAddChunk (_stream_readable.js:271:9)
	at Socket.Readable.push (_stream_readable.js:212:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:186:23) circuitpython.library.list

```

joedevivo, 2021-04-07:
Let me know if you still see this in v0.1.11

SAK917, 2021-04-07:
Sorry Joe, looks like it is still happening.  I tried a simple update of the extension to 0.1.11, then tried uninstalling and reinstalling in case that helped but no joy:

`
[2021-04-06 19:16:18.596] [exthost] [error] TypeError: Cannot read property 'name' of undefined
	at new LibraryQP (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\librarymanager\libraryManager.js:41:24)
	at c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\librarymanager\libraryManager.js:174:26
	at Array.forEach (<anonymous>)
	at LibraryManager.getInstalledChoices (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\librarymanager\libraryManager.js:171:59)
	at LibraryManager.<anonymous> (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\librarymanager\libraryManager.js:149:32)
	at Generator.next (<anonymous>)
	at c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\librarymanager\libraryManager.js:8:71
	at new Promise (<anonymous>)
	at __awaiter (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\librarymanager\libraryManager.js:4:12)
	at LibraryManager.list (c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\librarymanager\libraryManager.js:148:16)
	at c:\Users\SAK\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.11\out\container.js:52:77
	at S._executeContributedCommand (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:86:106770)
	at S.$executeContributedCommand (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:86:107208)
	at u._doInvokeHandler (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:12794)
	at u._invokeHandler (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:12478)
	at u._receiveRequest (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:11145)
	at u._receiveOneMessage (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:9931)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:90:8032
	at _.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:57:1836)
	at E.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:15496)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:105:30083
	at _.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:57:1836)
	at E.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:15496)
	at t._receiveMessage (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:20754)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:17640
	at _.fire (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:57:1836)
	at acceptChunk (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:12861)
	at c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:65:12209
	at Socket.w (c:\Users\SAK\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:105:13173)
	at Socket.emit (events.js:315:20)
	at addChunk (_stream_readable.js:295:12)
	at readableAddChunk (_stream_readable.js:271:9)
	at Socket.Readable.push (_stream_readable.js:212:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:186:23) circuitpython.library.list
`

Idirianuk, 2021-05-03:
By way of a bump on this one...

I'm seeing this too on Ubuntu 20.04, Windows 10 on x64 and Raspberry Pi OS on ARM.

Having looked at it and having managed to get the extension to build for myself (after several hours of head scratching: I know nothing of Typescript, tbh and have done little coding in 20 years) but it seems that 'bundlelib' isn't getting populated. The value "b: Library" in the constructor for the LibraryQP class is coming through as a NULL value.

I *think* that something isn't working when the "globby" library is used in the "loadLibraryMetadata" function and reads through the globalstorage directory enumerating the libraries stored there. It's not populating the array. This is certainly working when enumerating which libraries (in the /lib directory) are in use on the board being used as these are picked up. Somewhere in there there's an array full of "Promises". Again, ignorance of Typescript fails to help me here.

I can't add any more at this point because, as I said, I have no typescript experience and and not a practiced programmer currently, but I do hope this can be fixed because when it's working properly (and it has been for me) it's really useful. As it stands, I'm using Circup to help with libraries externally and Pico Go to provide a REPL console within VS.Code.

HTH. :)

alflanagan, 2021-11-28:
Probably redundant, but just in case: using code-insiders
```
Version: 1.63.0-insider
Commit: d18d093403b12a65350c58a7b0d5771cc1f42aba
Date: 2021-11-26T06:36:38.550Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.11.0-40-generic
```
on Ubuntu 21.04. The error is 
```
[2021-11-28 13:30:25.802] [exthost] [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onStartupFinished'
[2021-11-28 13:32:06.409] [exthost] [error] TypeError: Cannot read property 'name' of undefined
    at new LibraryQP (/home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:41:24)
    at /home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:175:26
    at Array.forEach (<anonymous>)
    at LibraryManager.getInstalledChoices (/home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:172:59)
    at LibraryManager.<anonymous> (/home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:150:32)
    at Generator.next (<anonymous>)
    at /home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:4:12)
    at LibraryManager.list (/home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:149:16)
    at /home/lloyd/.vscode-insiders/extensions/joedevivo.vscode-circuitpython-0.1.15/out/container.js:52:77
    at r._executeContributedCommand (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:95:31080)
    at r.$executeContributedCommand (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:95:31622)
    at t._doInvokeHandler (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:13802)
    at t._invokeHandler (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:13486)
    at t._receiveRequest (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:12148)
    at t._receiveOneMessage (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:10826)
    at /usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:99:8922
    at a.fire (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:66:1712)
    at r.fire (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:18929)
    at /usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:114:34324
    at a.fire (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:66:1712)
    at r.fire (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:18929)
    at n._receiveMessage (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:23321)
    at /usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:21005
    at a.fire (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:66:1712)
    at _.acceptChunk (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:15757)
    at /usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:74:14887
    at Socket.I (/usr/share/code-insiders/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:114:13813)
    at Socket.emit (events.js:315:20)
    at Socket.EventEmitter.emit (domain.js:467:12)
    at addChunk (internal/streams/readable.js:309:12)
    at readableAddChunk (internal/streams/readable.js:284:9)
    at Socket.Readable.push (internal/streams/readable.js:223:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:188:23) circuitpython.library.list {"value":"joedevivo.vscode-circuitpython","_lower":"joedevivo.vscode-circuitpython"}
```
board is an UnexepectedMaker Feathers2, using circuitpython 7.1.0-beta.0.