joedevivo/vscode-circuitpython #46: command 'circuitpython.openSerialMonitor' not found

Wintaru, 2021-04-12:
Command 'CircuitPython: Open Serial Monitor' resulted in an error (command 'circuitpython.openSerialMonitor' not found)

I tried with the build indicated here:

https://github.com/joedevivo/vscode-circuitpython/issues/38#issuecomment-814239981

with no luck. I get the above text on my M1 Mac running Big Sur.

Wintaru, 2021-04-12:
[image]

This happens in Windows 10 as well.

elexx, 2021-04-12:
I can reproduce this on Linux (Manjaro 21.0.1, Kernel 5.10.26-1). Downgrading the Extension to 0.1.8 resolves this problem for me.

wildestpixel, 2021-05-01:
This has now started to occur for myself - I cannot use the extension - only thing it now does is update the bundle and cannot do anything else. VSCode Universal binary 1.55.2 on M1 Big Sur 11.3. Seems to coincide with 11.3 update for me but only be coincidence - may be updates to MS python package or something else.

wildestpixel, 2021-05-01:
Same error trying to show available libraries : 

```
[2021-05-01 17:05:35.465] [exthost] [error] TypeError: Cannot read property 'name' of undefined
	at new LibraryQP (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:41:24)
	at /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:174:26
	at Array.forEach (<anonymous>)
	at LibraryManager.getInstalledChoices (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:171:59)
	at LibraryManager.getAllChoices (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:165:37)
	at LibraryManager.<anonymous> (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:140:32)
	at Generator.next (<anonymous>)
	at /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:8:71
	at new Promise (<anonymous>)
	at __awaiter (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:4:12)
	at LibraryManager.show (/Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/librarymanager/libraryManager.js:139:16)
	at /Users/pixel/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.11/out/container.js:51:77
	at S._executeContributedCommand (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:86:106770)
	at S.$executeContributedCommand (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:86:107208)
	at u._doInvokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:12794)
	at u._invokeHandler (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:12478)
	at u._receiveRequest (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:11145)
	at u._receiveOneMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:9931)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:90:8032
	at _.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:57:1836)
	at E.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:65:15496)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:29750
	at _.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:57:1836)
	at E.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:65:15496)
	at t._receiveMessage (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:65:20754)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:65:17640
	at _.fire (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:57:1836)
	at acceptChunk (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:65:12861)
	at /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:65:12209
	at Socket.w (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:105:13173)
	at Socket.emit (events.js:315:20)
	at addChunk (_stream_readable.js:295:12)
	at readableAddChunk (_stream_readable.js:271:9)
	at Socket.Readable.push (_stream_readable.js:212:10)
	at Pipe.onStreamRead (internal/stream_base_commons.js:186:23) circuitpython.library.show
```

kwalkerk, 2021-05-01:
Having the same problem on Ubuntu 20.04.  None of the circuitpython commands work.  Here is example error: Command 'CircuitPython: Open Serial Monitor' resulted in an error (command 'circuitpython.openSerialMonitor' not found)

ysoldak, 2021-05-04:
Does not work for me either, none of commands can be found.

My setup
- macOS Big Sur 11.3.1
- VSCode 1.55.2
- CircuitPython plugin 0.11.1
- Python plugin v2021.4.765268190
- Pylance plugin v2021.4.3

a-27m, 2021-05-04:
Disabling extension _JavaScript Debugger (ms-vscode.js-debug-nightly)_ solved the problem for me,

macOS Big Sur 11.3 (Intel)
VSCode 1.55.2
CircuitPython 0.1.11

GregWoods, 2021-05-09:
Getting the same issue on Windows
The only extensions enabled are...

CircuitPython 0.1.13
Jupyter 2021.6.811652604
PyLance 2021.5.1
Python 2021.4.765268190

I tried running as administrator as well. No joy.
All the CircuitPython extension commands fail.
I am happy to supply additional info, just let me know what is needed

dannystaple, 2021-05-16:
Reproduced on MacOS Big Sur, VSCode 1.56.1. Fresh install CircuitPython 0.1.13.
The extension Host log has an error similar to @GregWoods. I've added the full traceback here:

```
[2021-05-16 13:18:59.121] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-05-16 13:18:59.121] [exthost] [error] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at validateString (internal/validators.js:124:11)
	at Object.resolve (path.js:980:7)
	at ProviderAsync._getRootDirectory (/Users/danielstaple/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/providers/provider.js:17:21)
	at ProviderAsync.read (/Users/danielstaple/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/providers/async.js:11:27)
	at Array.map (<anonymous>)
	at getWorks (/Users/danielstaple/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/index.js:58:18)
	at FastGlob (/Users/danielstaple/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/index.js:10:19)
	at /Users/danielstaple/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/globby/index.js:133:52
	at Array.map (<anonymous>)
	at module.exports (/Users/danielstaple/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/globby/index.js:133:40)
```

This feels like some extension path configuration setting here. @cshields  may be on to something. I also found the null:

```
    "python.analysis.extraPaths": [
        null,
        "/Users/danielstaple/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/stubs",
        "/Users/danielstaple/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20210514/adafruit-circuitpython-bundle-py-20210514/lib"
    ]
```

Going to try remove it and see what happens.

dannystaple, 2021-05-16:
If I remove the null, and restart vscode, I see the same error, and the null has sprung back.

ejangi, 2021-06-09:
I was getting this error, but it turns out that the board only had the UF2 bootloader on it and I needed to install circuit python. Once I did that, the errors went away.

dannystaple, 2021-06-15:
Unfortunately - I'm not finding that. I have CircuitPython on there, I'm able to see the CIRCUITPY volume, copy code there.
The command `screen /dev/tty.usbmodem144101` gives me the repl so I am able to get serial comms with the board.

fivesixzero, 2021-08-05:
Ran into this problem today on a Windows system. Which is weird because on another Windows laptop the serial monitor has been working fine.

The rest of the extension seems to work fine - library management, board info, etc. No errors or odd behavior.

On trying to start the serial monitor, I get the following error.

>Command 'CircuitPython: Open Serial Monitor' resulted in an error (command 'circuitpython.openSerialMonitor' not found)

Tried the 'Close Serial Monitor' command and got a similar message:

>Command 'CircuitPython: Close Serial Monitor' resulted in an error (command 'circuitpython.closeSerialMonitor' not found)

And, for good measure, I tried 'Select Serial Port', which also resulted in a similar error:

>Command 'CircuitPython Select Serial Port' resulted in an error (command 'circuitpython.openSerialMonitor' not found)

None of the logs in the Outputs panel reported any errors during VS Code startup, extension install, or when the commands were run.

However, there was interseting stuff in the console output that might be helpful:

```logs
console.ts:137 [Extension Host] Error: Could not locate the bindings file. Tried:
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\build\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\build\Debug\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\build\Release\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\out\Debug\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\Debug\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\out\Release\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\Release\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\build\default\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\compiled\14.16.0\win32\x64\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\addon-build\release\install-root\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\addon-build\debug\install-root\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\addon-build\default\install-root\bindings.node
 → c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\lib\binding\node-v89-win32-x64\bindings.node
	at bindings (c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\bindings\bindings.js:126:9)
	at Object.<anonymous> (c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\lib\win32.js:1:98)
	at Module.u._compile (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:1315)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.n._load (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:104:28060)
	at Function.P._load (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:104:24648)
	at Function.l._load (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:94:59608)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at Module.require (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\microsoft-authentication\dist\extension.js:1:38760)
	at v (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:698)
	at Object.<anonymous> (c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\node_modules\@serialport\bindings\lib\index.js:6:22)
	at Module.u._compile (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:1315)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12913)
	at Function.n._load (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:104:28060)
	at Function.P._load (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:104:24648)
	at Function.l._load (c:\Users\me\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:94:59608)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
[... cut ...]
	at Container.<anonymous> (c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\out\container.js:59:17)
	at Generator.next (<anonymous>)
	at fulfilled (c:\Users\me\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.14\out\container.js:5:58)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
```

At first glance this seems to narrow things down to a problem loading the `serialport` node module. Unfortunately it looks like VSCode's extension host is eating the exception details though. :(

fivesixzero, 2021-08-05:
I ended up having some more time to dig today, and I think I found a fix, or at least something to pinpoint the issue. 🎉 

TL;DR: It looks like the `serialport` library that gets installed by default is compliled against a different version of the node ABI that VSCode's current version of Electron uses. So the library doesn't load, causing the enture `serial` module of the extension to fail to load.

My local fix required a bit of tinkering but I boiled it down to a few commands. It also spurred me to learn a bit about Node's package management, ABI versions, and Electron's bundled node versions, which was fun since my experience with those aspects of nodejs/electron is (so far) a bit limited.

Here are the commands I ran after installing the latest version of NodeJS:

```sh
rm /node_modules/serialport
rm /node_modules/@serialport

npm install electron-rebuild --save-dev
npm install -g node-abi
npm install serialport

./node_modules/.bin/electron_rebuild --version 13.1.7
```

After this was done I was able to launch VSCode and use the three serial-related commands without any problems. The developer tools console was clear errors as well. :)

I'm not familiar enough (yet) with Node/Electron ABI selection and how they relate to VSCode packages to offer up a PR so hopefully this info will help find a fix.

Wintaru, 2021-09-29:
> I ended up having some more time to dig today, and I think I found a fix, or at least something to pinpoint the issue. tada
> 
> TL;DR: It looks like the `serialport` library that gets installed by default is compliled against a different version of the node ABI that VSCode's current version of Electron uses. So the library doesn't load, causing the enture `serial` module of the extension to fail to load.
> 
> My local fix required a bit of tinkering but I boiled it down to a few commands. It also spurred me to learn a bit about Node's package management, ABI versions, and Electron's bundled node versions, which was fun since my experience with those aspects of nodejs/electron is (so far) a bit limited.
> 
> Here are the commands I ran after installing the latest version of NodeJS:
> 
> ```shell
> rm /node_modules/serialport
> rm /node_modules/@serialport
> 
> npm install electron-rebuild --save-dev
> npm instal node-abi
> npm install serialport
> 
> ./node_modules/.bin/electron_rebuild --version 13.1.7
> ```
> 
> After this was done I was able to launch VSCode and use the three serial-related commands without any problems. The developer tools console was clear errors as well. :)
> 
> I'm not familiar enough (yet) with Node/Electron ABI selection and how they relate to VSCode packages to offer up a PR so hopefully this info will help find a fix.
> 
> _edit: Got the idea to try this approach from this issue in the `serialport` repo: [serialport/node-serialport#1891](https://github.com/serialport/node-serialport/issues/1891)_

I must be missing something because I tried to run electron-rebuild and it can't find the version indicated.

fivesixzero, 2021-10-30:
> I must be missing something because I tried to run electron-rebuild and it can't find the version indicated.

Sorry my quick hack didn't work out. :(

The Node ABI version used by VSCode or the version used to complie the `serialport` lib may have changed since I originally wrote that reply. Additionally, I recall having some weird issues with getting the right `node-abi` version(s) installed/availble that, in retrospect, I wish past me had left notes about. 🙄 Could be some memory-wires crossed there though. So some more research and testing will be required to find a proper workaround and/or fix.

If I have time this weekend to take a closer look myself I'll dig in and update this reply.

cverbiest, 2021-12-23:
FWIW I had the same issue on Windows 10 while an update of Vscode was applied but VsCode was not yet restarted.
I'm my case restart VsCode was enough to fix the issue

Methodician, 2022-01-10:
Not exactly a solution to this problem, but a potential work-around that worked for me: https://learn.adafruit.com/welcome-to-circuitpython/advanced-serial-console-on-windows

JohnHind, 2022-01-27:
I originally saw this bug under Windows 10 and stopped using the extension because of it. I have since done a fresh OS install using Windows 11. Initially the extension worked fine, but the bug came back after I had installed some other extensions. I discovered a good work-around is simply to uninstall and re-install the extension. It seems to recur after I have been messing about with other extensions, but this may just be a coincidence.

Methodician's Solution 1 does not work for me because I have always been opening the CIRCUITPY drive as a folder and I still see the bug. Solution 2 rather defeats the point of the extension since it just uses VSCode as an editor, and the extension is not adding much. I found it easier just to use the Mu editor cutting and pasting code between it and VSCode for longer editing sessions.

HughGrant, 2022-03-07:
I'm on win11, all I do is uninstall and reinstall this plugin, and it worked.

omerk, 2022-03-14:
Windows 11, latest VS Code release. Uninstal and reinstall working here too did not work the first time around.

solarkraft, 2022-09-21:
That's not working for me with v0.1.19 installed on VSCodium 1.70.0 from Open VSX.

devoh747, 2022-09-21:
Just had the same error.  I was able to clear it with opening up the directory on the board (that is running circuitpython) and then selecting the file.  Uninstall/reinstall didn't work for me but the open directory did.

seanmrice, 2022-10-11:
> Just had the same error. I was able to clear it with opening up the directory on the board (that is running circuitpython) and then selecting the file(code.py). It puts a .vscode directory on the board with a settings.json file in it. Uninstall/reinstall didn't work for me but the open directory did.

Can confirm same resolution here.  Opened the directory instead of the code.py file and issue resolved.

ericmharris, 2022-11-26:
Ctrl+Shift+p (on Windows)
Workspaces: Add Folder to Workspace...
Add the directory/drive containing your devices project files (eg. boot_out.txt, code.py)
See  ...\extensions\joedevivo.vscode-circuitpython-0.1.19-win32-x64\out\librarymanager\libraryManager.js:getProjectRoot() for better understanding of problem.

ericmharris, 2022-11-26:
And really nice work on the extension ... thanks much!

bgoosman, 2023-01-03:
I ran into this today and narrowed it down to disabling Pylance extension 🤦🏻.

StephenFerrero, 2023-01-21:
> Just had the same error. I was able to clear it with opening up the directory on the board (that is running circuitpython) and then selecting the file(code.py). It puts a .vscode directory on the [...]

Using 'Open Folder' and selecting the CircuitPython drive worked to fix this for me on Win 10.

nfoert, 2023-07-11:
> Using 'Open Folder' and selecting the CircuitPython drive worked to fix this for me on Win 10.

Worked for me as well on Windows 11.

Patrick8195, 2024-01-07:
Found a solution to this problem on my **Macbook Pro M2 on Sonama 14.2.1** with VS Code and extension installed.

I needed to let the VS Code application have full access to the disk drive, then installed and reinstalled the extension. Works fine now.

Look under the **System Settings --> Privacy and Security --> Full Disk Access** and **turn it on** for VS Code. 

There probably is a way to restrict what directories VS Code needs, but I found this to work for me.

Hope this helps...

nicelifeBS, 2024-02-04:
The Open Folder solution doesn't work for me on Win10 with version 0.2.0 still get the error that the console cannot open.

mnai01, 2024-02-23:
On Win 10, downgrading extension to v0.2.0 worked for me

rossiniscarface, 2024-02-27:
Installed latest version of Visual Studio Code on Windows 11; Freshly installed CircuitPython extension v0.2.0 and none of the commands work, or the drive is recognized. Tried to install v0.1.20 and it works?

sveinungkb, 2024-03-01:
Rolling back worked here too. Ubuntu 22.

KineticPrecision, 2024-03-13:
Downgrading CircuitPython extension to 0.1.20 also worked here.  Win 10.

johnwyles, 2024-03-15:
For those wondering use these instructions to bump down to 0.1.20 (which is all that fixed this for me): https://bobbyhadz.com/blog/install-older-version-of-extension-in-vscode

sidprice, 2024-03-23:
Rolling back to 0.1.20 worked on Windows too.

stoporjinschi, 2024-05-18:
This problem is still reproducible .... for me at least... 
Current instalation:
Windows 11 (fresh install) 
vsCode (portable latest version) 
extensions installed
[image]
I tried everything above, nothing worked, for all circuitPython extension all commands are not found.....

I can see that first occurence is 3 years old  so fair question ... is this extension still maintained ???

SergiuToporjinschi, 2024-05-18:
This problem is still reproducible .... for me at least... 
Current instalation:
Windows 11 (fresh install) 
vsCode (portable latest version) 
extensions installed
[image]
I tried everything above, nothing worked, for all circuitPython extension all commands are not found.....

I can see that first occurence is 3 years old  so fair question ... is this extension still maintained ???

SuarezLuis, 2024-05-30:
Mac M1 Ultra, rolling back to 1.20 worked for me

epiqu1n, 2024-05-31:
I am also experiencing this issue. I was really hoping this would work as I do not like Mu... guess I'll go back to Thonny

JohnHind, 2024-05-31:
> I am also experiencing this issue. I was really hoping this would work as I do not like Mu... guess I'll go back to Thonny

That is a bit drastic! I use the Microsoft 'Serial Monitor' plugin in conjunction with the native ability to open the code directly on the device with the MSD implementation in CircuitPython. The only other thing the CircuitPython plugin did was a little light library management which is pretty easy to do manually, or directly from GitHub using the built in integration.

epiqu1n, 2024-06-01:
Oh! Thank you that works great; I was a bit tired that night so a bit too defeatist I guess 😅

johnwyles, 2024-06-01:
> Oh! Thank you that works great; I was a bit tired that night so a bit too defeatist I guess 😅

I hear ya. With kubernetes or homeassistant it’s usually 🙂 -> 😕 -> 🥴 -> 🙃 … 🤣

Patcybermind, 2024-07-10:
same problem here 1.20 works but none other versions seem to work

CombatVolcano, 2024-07-26:
same problem what gives this is pretty basic and important

trolleyman, 2024-11-18:
Also have this issue :/