joedevivo/vscode-circuitpython #48: [BUG] Update to VSC caused a problem loading bindings file

wmerkens, 2021-05-06:
**Describe the bug**
When VSC attempts to load the extension it fails with a problem finding a bindings file

**To Reproduce**
Steps to reproduce the behaviour:
1. Start VSC
2. Open a Circuitpython CIRCUITPY folder and open code.py
3. Look at bottom toolbar to the right and no board selection  is offered
4. Open the LOG (Extension Host)

**Expected behavior**
Should be able to pick a board

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 (KDE Neon)
 - Version VSC
    Version: 1.56.0
    Commit: cfa2e218100323074ac1948c885448fdf4de2a7f
    Date: 2021-05-04T21:58:14.757Z
    Electron: 12.0.4
    Chrome: 89.0.4389.114
    Node.js: 14.16.0
    V8: 8.9.255.24-electron.0
    OS: Linux x64 5.4.0-72-generic

**Additional context**
These lines in the log show off the problem with bindings

wmerkens, 2021-05-06:
From what I can figure out not being a npm/node.js person the version of electron being used needs to b V 12.0.6 so that node-v87 is used. I tried to update it but was unable since I don't do node.js stuff and cannot figure out what I am doing wrong.

joedevivo, 2021-05-07:
Yeah, this happens every time they update the major version of electron. I'll take a look this evening, and hopefully figure out how to future proof this at least a little.

wmerkens, 2021-05-08:
Ok put VSC back to 1.56 installed 0.1.12 update I get this error

`[2021-05-08 11:42:10.041] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-05-08 11:42:10.041] [exthost] [error] Error: Could not locate the bindings file. Tried:
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/build/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/build/Release/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/Debug/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/out/Release/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/Release/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/build/default/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/compiled/14.16.0/linux/x64/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/lib/binding/node-v87-linux-x64/bindings.node
	at bindings (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/bindings/bindings.js:126:9)
	at Object.<anonymous> (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/lib/linux.js:2:36)
	at Module.u._compile (/usr/share/code/resources/app/out/vs/loader.js:4:1311)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12684)
	at Function.s._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:106:27816)
	at Function.S._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:106:24532)
	at Function.a._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:96:58822)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at v (/usr/share/code/resources/app/out/vs/loader.js:4:694)
	at Object.<anonymous> (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/@serialport/bindings/lib/index.js:14:22)
	at Module.u._compile (/usr/share/code/resources/app/out/vs/loader.js:4:1311)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12684)
	at Function.s._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:106:27816)
	at Function.S._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:106:24532)
	at Function.a._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:96:58822)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at v (/usr/share/code/resources/app/out/vs/loader.js:4:694)
	at Object.<anonymous> (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/node_modules/serialport/lib/index.js:2:17)
	at Module.u._compile (/usr/share/code/resources/app/out/vs/loader.js:4:1311)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
	at Module._load (internal/modules/cjs/loader.js:823:14)
	at Function.f._load (electron/js2c/asar_bundle.js:5:12684)
	at Function.s._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:106:27816)
	at Function.S._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:106:24532)
	at Function.a._load (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:96:58822)
	at Module.require (internal/modules/cjs/loader.js:1006:19)
	at v (/usr/share/code/resources/app/out/vs/loader.js:4:694)
	at Object.<anonymous> (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.12/out/serialmonitor/serialMonitor.js:15:20)
	at Module.u._compile (/usr/share/code/resources/app/out/vs/loader.js:4:1311)
	at Object.Module._extensions..js (internal/modules/cjs/loader.js:1155:10)
	at Module.load (internal/modules/cjs/loader.js:982:32)
[... cut ...]
	at d._loadCommonJSModule (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:106:25941)
	at d._doActivateExtension (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:92:12457)
	at d._activateExtension (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:92:11516)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
	at async Promise.all (index 0)
	at async Promise.all (index 90)
	at async Promise.all (index 1)`

joedevivo, 2021-05-08:
Give [this](https://artprodeus21.artifacts.visualstudio.com/Ad3f6d785-5e77-4982-8d31-52d4c13c0240/9b023c72-72b6-4ba6-86e5-86bb7df66b40/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2pvZWRldml2by9wcm9qZWN0SWQvOWIwMjNjNzItNzJiNi00YmE2LTg2ZTUtODZiYjdkZjY2YjQwL2J1aWxkSWQvODEvYXJ0aWZhY3ROYW1lL3BhY2thZ2U1/content?format=file&subPath=%2Fvscode-circuitpython-dev-c9b7325a.vsix) build a try, I just tested it on my M1, so it should be good.

wildestpixel, 2021-05-08:
That works - I can finally use to update libs again on device - no more using circus manually in separate terminal - Many thanks Joe

dmlond, 2021-05-08:
thanks so much. Using that build worked. I assume that an official update will be released that I can/should install soon?

wildestpixel, 2021-05-08:
Its good to be back on latest VS COde too instead of rolling back to 1.55, rolling back Jupiter, Pylance and Python manually to get things working - thanks Joe for the great support of this plugin / extension

joedevivo, 2021-05-08:
0.1.13 is building now, will probably be available in the marketplace within an hour

wmerkens, 2021-05-08:
Ok just went back to VSC 1.56 and it updated to 0.1.13 curcuitpython and everything is working correctly so far. Thanks for the fix.

closed 2021-05-10 by joedevivo