joedevivo/vscode-circuitpython #50: [BUG] Cannot pick a board

wmerkens, 2021-05-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the bottom right in VSC and pick a board
2. Click on <Choose a board>
3. Error Message "command 'circuitpython.selectBoard' not found"
4. See the error

**Expected behavior**
Should bring up the board list

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
Version: 1.56.2
Commit: 054a9295330880ed74ceaedda236253b4f39a335
Date: 2021-05-12T16:45:26.313Z
Electron: 12.0.4
Chrome: 89.0.4389.114
Node.js: 14.16.0
V8: 8.9.255.24-electron.0
OS: Linux x64 5.4.0-72-generic

**Additional context**

`[2021-05-21 02:07:06.660] [exthost] [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
[2021-05-21 02:07:06.660] [exthost] [error] TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at validateString (internal/validators.js:124:11)
	at Object.resolve (path.js:980:7)
	at ProviderAsync._getRootDirectory (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/providers/provider.js:17:21)
	at ProviderAsync.read (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/providers/async.js:11:27)
	at Array.map (<anonymous>)
	at getWorks (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/index.js:58:18)
	at FastGlob (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/fast-glob/out/index.js:10:19)
	at /home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/globby/index.js:133:52
	at Array.map (<anonymous>)
	at module.exports (/home/willm/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/node_modules/globby/index.js:133:40)`

The extension seems to work but you cannot pick a board

wmerkens, 2021-05-21:
A clarification, after writing about this I realized by looking at the pull request it might be an unrecognized board so I tested again, and yes it's not recognizing an adafruit itsy bitsy board but when I re-tested against an RP Pico plus the Trinket M0 there was no problem.

wmerkens, 2021-05-21:
Not sure how board detection works but here is the details of the ItsyBitsy as seen from Linux dmesg

`[276555.316313] usb 2-2.3: new full-speed USB device number 25 using uhci_hcd
[276555.566383] usb 2-2.3: New USB device found, idVendor=239a, idProduct=802c, bcdDevice= 1.00
[276555.566385] usb 2-2.3: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[276555.566386] usb 2-2.3: Product: ItsyBitsy M4 Express
[276555.566387] usb 2-2.3: Manufacturer: Adafruit Industries LLC
[276555.566387] usb 2-2.3: SerialNumber: FF21C670359333350202026312C290FF
[276555.576442] cdc_acm 2-2.3:1.0: ttyACM0: USB ACM device
[276555.589445] usb-storage 2-2.3:1.2: USB Mass Storage device detected
[276555.589741] scsi host33: usb-storage 2-2.3:1.2
[276555.605174] input: Adafruit Industries LLC ItsyBitsy M4 Express Keyboard as /devices/pci0000:00/0000:00:11.0/0000:02:00.0/usb2/2-2/2-2.3/2-2.3:1.3/0003:239A:802C.0015/input/input46
[276555.664217] input: Adafruit Industries LLC ItsyBitsy M4 Express Mouse as /devices/pci0000:00/0000:00:11.0/0000:02:00.0/usb2/2-2/2-2.3/2-2.3:1.3/0003:239A:802C.0015/input/input47
[276555.664424] input: Adafruit Industries LLC ItsyBitsy M4 Express Consumer Control as /devices/pci0000:00/0000:00:11.0/0000:02:00.0/usb2/2-2/2-2.3/2-2.3:1.3/0003:239A:802C.0015/input/input48
[276555.664497] input: Adafruit Industries LLC ItsyBitsy M4 Express as /devices/pci0000:00/0000:00:11.0/0000:02:00.0/usb2/2-2/2-2.3/2-2.3:1.3/0003:239A:802C.0015/input/input49
[276555.664867] hid-generic 0003:239A:802C.0015: input,hidraw2: USB HID v1.11 Keyboard [Adafruit Industries LLC ItsyBitsy M4 Express] on usb-0000:02:00.0-2.3/input3
[276556.627655] scsi host33: scsi scan: INQUIRY result too short (5), using 36
[276556.627659] scsi 33:0:0:0: Direct-Access     Adafruit ItsyBitsy M4 Exp 1.0  PQ: 0 ANSI: 2
[276556.632210] sd 33:0:0:0: Attached scsi generic sg2 type 0
[276556.632679] sd 33:0:0:0: [sdb] 4089 512-byte logical blocks: (2.09 MB/2.00 MiB)
[276556.635731] sd 33:0:0:0: [sdb] Write Protect is off
[276556.635733] sd 33:0:0:0: [sdb] Mode Sense: 03 00 00 00
[276556.638662] sd 33:0:0:0: [sdb] No Caching mode page found
[276556.638664] sd 33:0:0:0: [sdb] Assuming drive cache: write through
[276556.685723]  sdb: sdb1
[276556.705724] sd 33:0:0:0: [sdb] Attached SCSI removable disk`

If this helps to amend the board list

EricPobot, 2021-05-23:
This extension seems to be buggy. Or maybe it's something in VSCode which is.

I just installed it on a fresh VSCode on a laptop recently configured as a Mint 20 box (my other one being in Mint 19, the extension complained about requiring glibc 2.18+ :/), and all its commands fail with a message "command 'circuitpython.xxx' not found.

Anyways, since the commands it brings are nothing that cannot be done easily by hand, I think that I'll ditch it for the moment. 

As a workaround: 
- in the workspace settings JSON file, add `<home_path>/.vscode/extensions/jeodevivo..../boards/0x239A/0x80FA/` to the key `settings/python.analysis.extraPaths` (a Pylance setting) to avoid `unresolved xxx` errors notified on `import` statement for instance and to benefit from auto-completion
- as a replacement of the "Open Serial Monitor" use any terminal emulator (my preference is picocom, but anything else will work) and connect it to `/dev/ttyACM0` with speed set to 115200 (the serial port path may vary but on a Linux box it will always be `ttyACMx`)
- save the Python files directly on the pseudo-disk (i.e. /media/<username>/CIRCUITPY) to "upload" (in Arduino IDE's parlance) them. A safer option (IMHO) is to copy them to this destination (using `cp` or `rsync` when it's time to deploy them on the Pico), otherwise they will never be saved on your machine and you'll have no copy if ever you're forced to nuke the Pico flash for whatever reason). 

A basic Makefile can be handy to gather all these commands in a single place and activate them easily. Since VSCode know how to deal with them, they will accessible from there if you prefer this way of working.

Once this is done, you can disable the extension. Do not uninstall it in order to keep the Python stub files for the imported modules. Or retrieve the stubs from elsewhere.

**Note:** where do the `0x...` parts of the path used above come from ? 
They are nothing else that the USB vendor and product standard identifiers. Use `lsub` to display the USB devices currently hooked to the system, and you'll find the Pico listed there with these ids. It can also be found in the `metadata.json` file located in the boards sub-folder of the extension. You'll need to format it (with `jq` for instance) to be able to locate the Pico since it's stored in compact form. The `lsusb` way if far simpler and quicker IMHO.

wmerkens, 2021-05-23:
@EricPobot the extension depends on electron which probably caused the Glibc dependencies error because the electron in use jumped versions this month from 1.55 to the 1.56 VSC. What this extension is supposed to do is help with code completion, circuit python has a set of python code as modules that are in the uf2 image you place on the circuit python board which IntelliSense does not see via the Pylance extension. If you want IntelliSense to work and especially Pylance then it needs some method to find them and the stubs that the extension provides fills this need. Pylance is very picky and circuit python has three types of imports that will cause grief, the stubs, mpy, and regular py files. As far as the code being on only the CIRCUITPY flash storage use git to store a copy. Also, the whole point of these boards is as soon as you save changes to code.py it is run, this won't happen if you code outside that folder. I also found I had to install via pip3 a bunch of extra Adafruit python stuff, blinka being a big one to get things working good in VSC.

EricPobot, 2021-05-23:
@wmerkens for the comment.

I suspect too that the culprit is VSCode, since the other Pico related extension I've tried had the same problem related to glibc. However, as I reported, I made a VSCode fresh install on a machine equipped with the proper glibc, and the error is not then glibc version anymore but that methods called by the various extension commands are not found. Quite strange knowing the I've installed just now, using the official procedure. So it looks like VSCode is not the culprit anymore.

WRT intellisense, I could "fix" the problem as I described, by putting the stubs (extracted from the archive available on Adafruit Web site) in a location added to the Pylance search path and everything is working fine now (symbol resolution, completion,...). 

So the features brought by the extensions as of now are opening the serial monitor and sending the Python files to the Pico simulated storage. Both of them are nothing more complicated that launching a terminal emulator for the first and doing a copy or a rsync for the second, which I packaged in a less that 10 lines (blank included) Makefile. Hence my interrogation about the concrete added value of these extensions. Which advanced feature do they offer ? Do they let you debug code running on the Pico for instance ? I didn't find a mention to this kind of feature yet, but have not searched in depth.

wmerkens, 2021-08-11:
Solved the issue with the missing board (and other import issues)

First I cloned from git this extension

I Learned how to rebuild the boards and stubs directory using the build-stubs.sh in the scripts folder

The only issue that was needed was to fix the script

```
#!/usr/bin/bash
rm -rf stubs        # these three lines are there if you ran the script already, cleans up first.
rm -rf boards
rm -rf circuitpython

git clone --depth 1 --branch main https://github.com/adafruit/circuitpython.git       # good
cd circuitpython      # good
git submodule init      # good
git submodule update extmod/ulab       # good
pip3 install -r docs/requirements.txt      # good
make stubs       # good
mv circuitpython-stubs ../stubs      # mv folder one directory up and rename it

cd ..      # cd up to ~/scripts  At this point this script starts to fail
cd ..      # need to go one level higher

pwd     # a debug just to confirm where you are, not needed.

pip3 install -r requirements.txt     # problem 1: this exists one directory higher, fixed by previous "cd .. cd .."
cd scripts     # go back into scripts so we can do the py script

# python3 ./scripts/build_stubs.py       # problem 2: this fails because you are in the wrong folder relative to the code in the python 
                                         # file, it's expecting to be in scripts but you are one up if you did not cd back into scripts
python3 build_stubs.py                   # fix it by running it like this after returning to the scripts directory.

#rm -rf stubs/board               # this should not be erased, causes an import problem.
```

once you run the script you will have two new subdirectories, stubs, and boards.

I would first go to 
```
~/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15

```
and move stubs and boards out of the way, mv stubs to stubs.org for example. repeat for boards

Then go back to where you have the new folders and copy or move them to that folder in .vscode

restart VSC and all should be working.