joedevivo/vscode-circuitpython #51: [BUG] Need to add a new board

wmerkens, 2021-07-04:
Want to add the new Arduino Nano RP2040 Connect
Product ID: 0x2341
Vendor ID: 0x005E

https://learn.adafruit.com/circuitpython-on-the-arduino-nano-rp2040-connect/overview

I created the directories

.vscode/extensions/joedevivo.vscode-circuitpython-0.1.13/boards/0x2341/0x005E/

And I placed a board.pyi file based on another Arduino and modified it.

```
from typing import Any
"""
board Arduino Arduino_Nano_RP2040_Connect
https://circuitpython.org/boards/arduino_nano_rp2040_connect
"""
```

now I am up to the pin list, is this generated manually or automatically?

rpavlik, 2021-08-16:
All this content is generated automatically. As long as it's in upstream CircuitPython, it will be included in the release next time the stubs are built and the extension is released.

wmerkens, 2021-08-19:
Thanks, I figured out the script for generating the boards and stubs directory and I fixed the issue.

piit79, 2022-11-02:
Can this be closed please? :) Thanks!

wmerkens, 2022-11-02:
Yeah, let's close it; bit old. :-)

closed 2022-11-02 by wmerkens