joedevivo/vscode-circuitpython #57: [BUG] pylance leads to cannot use circuitpython, need to remove dependency

DC-Shi, 2021-09-01:
**Describe the bug**
Cannot use circuitpython after window loaded, this because pylance blocked python plugin loading. Could you please remove dependency of pylance?

**To Reproduce**
Steps to reproduce the behavior:
1. The vscode without any python related plugin.
2. Install circuitpython, it will automatically install python(and pylance, jupyter)
3. Reload the window as indicated.
4. Stuck at "Python extension is loading..." and cannot use any circuitpython functions. 
[image]

**Expected behavior**
Run the circuitpython env and check for board.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version vscode 1.59.1, circuitpython 0.1.15

**Additional context**
I found the problem is pylance related because I did these tests:
- directly open python file with python extension installed(automatically installed pylance, jupyter), reload the window, python extension stucked.
- If I remove pylance only(MS says this is optional for python), and open python file, everything worked as expected.
- If I install circuitpython, due to it depends python, it will install pylance, jupyter as well. After reloading window, python extension stucked.
- If I want to remove pylance only, vscode told me circuitpython depends on it and I cannot remove pylance.
[image]
- If I just removed pylance, reload the window, install circuitpython, did NOT reload the window, I can finally use circuitpython. But next time when I closed the folder/application, I will encounter the stuck again.

MattHeffNT, 2021-09-29:
Same issue here with Windows, seems to work fine in Ubuntu though.

danyeaw, 2023-10-03:
According to the [Changelog](https://github.com/joedevivo/vscode-circuitpython/blob/master/CHANGELOG.md) this was fixed 0.1.18, can we close this issue?