joedevivo/vscode-circuitpython #61: New release?

rpavlik, 2021-09-28:
There have been a number of new boards since the last release, and there's also my improvements to the stubs that have been merged but not released. Would be great to get a new release out. Let me know how I can help.

wmerkens, 2021-10-11:
Thanks for the re-write of the builds-stubs.sh, one thing I noticed is that it's not exiting from the python venv.

    # get out of venv
    deactivate

It complains about deactivate not being found. I had to comment it out and replace it with "exit"

Also, why do you
rm -rf stubs/board

I find that causes issues in code for detection of the board module.

joedevivo, 2022-03-26:
I actually just had to remove the `deactivate` when moving to GitHub Actions, and confirmed that it is unnecessary because of the sub-shell. This is all in 0.1.16. Thanks!

closed 2022-03-26 by joedevivo