joedevivo/vscode-circuitpython #70: Cannot get it to work on Codium, Linux

Sciss, 2022-01-05:
I installed Codium via recommended [package manager](https://vscodium.com/#install). It's version 1.63.2; system is Debian 11; Python is 3.9.2. I think the marketplace doesn't work with Codium? Finding the extension from within Codium doesn't work. So I download the extension from the website, and install using `codium --install-extension joedevivo.vscode-circuitpython-0.1.15.vsix`.

But I think it's missing a step. I get a notification `Downloading new bundle: 20220105`, but nothing seems to happen. When I try to execute any CircuitPython command, say `CircuitPython: Select Serial Port` , I just get the error `command 'circuitpython.selectBoard' not found`.

Screenshot attached.

[image]

Sciss, 2022-01-05:
I quit, fixed the marketplace links like this: https://github.com/VSCodium/vscodium/issues/418#issuecomment-643664182

Restarted Codium, now it said 'Bundle already at latest version: 20220105', no more errors so far, 'Select Serial Port' opens selector.

closed 2022-01-05 by Sciss

mwaterbu, 2022-06-25:
> I quit, fixed the marketplace links like this: [VSCodium/vscodium#418 (comment)](https://github.com/VSCodium/vscodium/issues/418#issuecomment-643664182)

This didn't work for me. I have tried installing with the vsix file, and through the market (after doing the fix you suggested), neither way works. How is your Codium installed? I am running the Snap package.

It works fine for me with Codium on Windows when installing through the market.

On Linux, I cannot execute any CircuitPython commands, they all give the not found error. Other extensions from the marketplace work fine.
[image]

Anonymous941, 2023-11-27:
Exact same issue here.  I'd prefer not to switch to VSCode but it looks like I have to