joedevivo/vscode-circuitpython #72: Extension Looking for Incorrect 7.x Bundle

mattbeiswenger, 2022-01-19:
When attempting to install the `adafruit_esp32spi` library via the `Show Available Libraries` command, I get the following error:
```
[2022-01-18 22:43:31.587] [exthost] [error] Error: ENOENT: no such file or directory, stat '/Users/mattbeiswenger/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20220118/adafruit-circuitpython-bundle-7.x-mpy-20220118/lib/adafruit_esp32spi'
    at statSync (fs.js:1086:3)
    at Object.e.statSync (electron/js2c/asar_bundle.js:5:4315)
    at Object.statSync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/graceful-fs/polyfills.js:311:34)
    at getStatsSync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/fs-extra/lib/util/stat.js:58:18)
    at Object.checkPathsSync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/fs-extra/lib/util/stat.js:90:33)
    at Object.copySync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/fs-extra/lib/copy-sync/copy-sync.js:24:38)
    at LibraryQP.install (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:71:22)
    at LibraryQP.onClick (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:58:22)
    at LibraryManager.<anonymous> (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:144:24)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:5:58)
    at processTicksAndRejections (internal/process/task_queues.js:93:5) circuitpython.library.show {"value":"joedevivo.vscode-circuitpython","_lower":"joedevivo.vscode-circuitpython"}
[2022-01-18 22:45:10.747] [exthost] [error] Error: ENOENT: no such file or directory, stat '/Users/mattbeiswenger/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20220118/adafruit-circuitpython-bundle-7.x-mpy-20220118/lib/adafruit_esp32spi'
    at statSync (fs.js:1086:3)
    at Object.e.statSync (electron/js2c/asar_bundle.js:5:4315)
    at Object.statSync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/graceful-fs/polyfills.js:311:34)
    at getStatsSync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/fs-extra/lib/util/stat.js:58:18)
    at Object.checkPathsSync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/fs-extra/lib/util/stat.js:90:33)
    at Object.copySync (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/node_modules/fs-extra/lib/copy-sync/copy-sync.js:24:38)
    at LibraryQP.install (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:71:22)
    at LibraryQP.onClick (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:58:22)
    at LibraryManager.<anonymous> (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:144:24)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/mattbeiswenger/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.15/out/librarymanager/libraryManager.js:5:58)
    at processTicksAndRejections (internal/process/task_queues.js:93:5) circuitpython.library.show {"value":"joedevivo.vscode-circuitpython","_lower":"joedevivo.vscode-circuitpython"}
```

The installed bundle at the `/Users/mattbeiswenger/Library/Application Support/Code/User/globalStorage` location, however, is named `adafruit-circuitpython-bundle-py-20220118` and not `adafruit-circuitpython-bundle-7.x-mpy-20220118`. I've tried uninstalling and re-installing the plugin thinking that it would re-install the bundle correctly but that didn't help. I've also tried the `Check for latest bundle` command but it notifies me that `Bundle already at latest version: 20220118`.

If it helps, the boot_out.txt file states that my board is using CircuitPython 7.1.1:
```
Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
```

**Desktop**
MacOS Monterey 12.1 on Apple Silicon

laust, 2022-01-21:
I see the same problem. 

You can work around it (until the next update of CircuitPython libraries), by manually downloading the latest (mpy) bundle from the CircuitPython Libraries site, unpack and place the unpacked directory in the folder mentioned in the "no such file or directory" error message.

For the above, this would involve
- Download latest bundle (check date is the same as in the no such file or directory) error.
- Unpack, this should give you a directory with libraries with the name: adafruit-circuitpython-bundle-7.x-mpy-20220118
- Place unpacked directory in /Users/mattbeiswenger/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20220118/
- Now you can install libraries from within VS Code again.
- The date will change whenever the library bundle is updated and you will have to do it over again.

fivesixzero, 2022-01-22:
I ran into this problem as well today and the workaround mentioned by @laust worked perfectly. Thanks for sharing that deceptively simple fix! :)

HughGrant, 2022-03-08:
> I see the same problem.
> 
> You can work around it (until the next update of CircuitPython libraries), by manually downloading the latest (mpy) bundle from the CircuitPython Libraries site, unpack and place the unpacked directory in the folder mentioned in the "no such file or directory" error message.
> 
> For the above, this would involve
> 
> * Download latest bundle (check date is the same as in the no such file or directory) error.
> * Unpack, this should give you a directory with libraries with the name: adafruit-circuitpython-bundle-7.x-mpy-20220118
> * Place unpacked directory in /Users/mattbeiswenger/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython/bundle/20220118/
> * Now you can install libraries from within VS Code again.
> * The date will change whenever the library bundle is updated and you will have to do it over again.
> 
> P.S. I see this in VS Code 1.62.2 on MacOS 12.1 and CircuitPython plugin is the latest v 0.1.15

God, you're the best

joedevivo, 2022-03-13:
I found the error this morning. Once Adafruit stopped publishing `6.x-mpy` bundles, the extension still tried to download them, failed when they couldn't be found, and then never moved on to `7.x-mpy`. I've got a fix working locally and it will be included in an update today. Sorry about the inconvenience.

joedevivo, 2022-03-15:
I still need to test the build, but it's [here](https://artprodeus21.artifacts.visualstudio.com/Ad3f6d785-5e77-4982-8d31-52d4c13c0240/9b023c72-72b6-4ba6-86e5-86bb7df66b40/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2pvZWRldml2by9wcm9qZWN0SWQvOWIwMjNjNzItNzJiNi00YmE2LTg2ZTUtODZiYjdkZjY2YjQwL2J1aWxkSWQvMTE1L2FydGlmYWN0TmFtZS9wYWNrYWdl0/content?format=zip) if you want early access

joedevivo, 2022-03-22:
Finally was able to fix the build issues. There are platform specific builds here that should solve the problem: https://github.com/joedevivo/vscode-circuitpython/actions/runs/2022879323

commit e8c673c2 (2022-03-13): Fix #72
commit a393f6d8 (2022-03-13): Fix #72
commit 735db768 (2022-03-13): Fix #72
commit 09ab3a1e (2022-03-13): Fix #72
closed 2022-03-23 by joedevivo