joedevivo/vscode-circuitpython #75: [BUG] Error while installing libs

Karolis92, 2022-02-14:
**Describe the bug**
I get error when running:
2022-02-14 18:18:32.505] [exthost] [error] Error: ENOENT: no such file or directory, copyfile 'c:\Users\user\AppData\Roaming\Code\User\globalStorage\joedevivo.vscode-circuitpython\bundle\20220214\adafruit-circuitpython-bundle-7.x-mpy-20220214\lib\adafruit_24lc32.mpy' -> 'e:\lib\adafruit_24lc32.mpy'
	at Object.copyFileSync (fs.js:1991:3)
``

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kk-sonata, 2022-02-20:
The extension seems to be downloading the source bundle (.py) rather than the usual compiled (.mpy) one.

jbliesener, 2022-03-01:
Same here, Windows 11, VSCode 1.64.2, Extension v0.1.15

ghayne, 2022-03-06:
Same here on Linux. I always find it incredibly sad when developers create something as useful as this and then lose interest, for whatever reason. Such a pity, I'm sure the fix is not a great deal of work. I have a workaround using node-red to watch the bundle directory and download the correct library and save it as soon as the bundle date changes..

JanGerene, 2022-03-07:
Same problem.  Using Windows 11.  VSC 1.65.0 Extension v0.1.15

HughGrant, 2022-03-08:
same issue here, just like you.

> Same problem. Using Windows 11. VSC 1.65.0 Extension v0.1.15