joedevivo/vscode-circuitpython #84: [BUG] No Command Palette commands work, all give the same error (Ubuntu 20.04LTS)

han-fastolfe, 2022-03-30:
**Describe the bug**
All attempts to use any command palette commands give the following error:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Open VS Code with `code .` in the folder for a CIRCUITPY USB drive
2. Press F1 to open the Command Palette
3. Type `Circuit` to see the ones available to CircuitPython extension
4. See error

**Expected behavior**
That the command palette options work.

**Desktop (please complete the following information):**
VSCode Version: 1.65.2
Commit: c722ca6c7eed3d7987c0d5c3df5c45f6b15e77d1
Date: 2022-03-10T15:36:26.048Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Linux x64 5.14.0-1031-oem snap (Ubuntu 20.04.4 LTS)

SirJamesBrown, 2022-05-12:
Same issue

Version: 1.67.0
Commit: 57fd6d0195bb9b9d1b49f6da5db789060795de47
Date: 2022-05-04T12:08:42.978Z
Electron: 17.4.1
Chromium: 98.0.4758.141
Node.js: 16.13.0
V8: 9.8.177.13-electron.0
OS: Linux x64 5.13.0-41-generic

aschokking, 2022-07-15:
I'm getting this same error on windows 10

NikilXYZ, 2022-07-30:
Same, I'm on Mac OS Monterey 12.4

nucklearproject, 2023-07-17:
Same issue error here... 
Linux mint Cinamon
[image]