joedevivo/vscode-circuitpython #89: [BUG] CircuitPython: Show Available Libraries results in 'Cannot read properties of undefined (reading 'name')' error

alexlinde, 2022-04-24:
Was working when first installed. Uninstalled and reinstalled the CircuitPython extension, deleted the /Users/<<user>>/Library/Application Support/Code/User/globalStorage/joedevivo.vscode-circuitpython folder for good measure. Running the 'Show Available Libraries' command always results in this error message

MacOS 12.3.1
VSCode 1.66.2
CircuitPython v0.1.17

[image]

HughGrant, 2022-04-26:
same here, have you find any clue about it?

HughGrant, 2022-04-26:
@alexlinde currently, we can just use CIRCUP to manage the bundle library until this get fixed.

leeroywking, 2022-06-20:
This just started happening to me too for all the library related commands I tried. 
Circuit Python: List Project Libraries
Circuit Python: Show available libraries

tyeth, 2022-08-31:
Been happening here for a while, thought it was my fault, I did manually add non-standard micropython libraries.
The message is 
```
[2022-08-31 14:54:04.595] [exthost] [error] TypeError: Cannot read properties of undefined (reading 'name')
    at new LibraryQP (c:\Users\tyeth\.vscode-insiders\extensions\joedevivo.vscode-circuitpython-0.1.19-win32-x64\out\librarymanager\libraryManager.js:25:24)
    at c:\Users\tyeth\.vscode-insiders\extensions\joedevivo.vscode-circuitpython-0.1.19-win32-x64\out\librarymanager\libraryManager.js:149:26
    at Array.forEach (<anonymous>)
    at LibraryManager.getInstalledChoices (c:\Users\tyeth\.vscode-insiders\extensions\joedevivo.vscode-circuitpython-0.1.19-win32-x64\out\librarymanager\libraryManager.js:146:59)
    at LibraryManager.update (c:\Users\tyeth\.vscode-insiders\extensions\joedevivo.vscode-circuitpython-0.1.19-win32-x64\out\librarymanager\libraryManager.js:134:28)
    at c:\Users\tyeth\.vscode-insiders\extensions\joedevivo.vscode-circuitpython-0.1.19-win32-x64\out\container.js:41:75
    at c._executeContributedCommand (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:85:63770)
    at c.$executeContributedCommand (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:85:64474)
    at i._doInvokeHandler (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:95:13689)
    at i._invokeHandler (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:95:13371)
    at i._receiveRequest (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:95:12141)
    at i._receiveOneMessage (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:95:10843)
    at c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:95:8950
    at b.invoke (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:61:145)
    at p.deliver (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:61:2266)
    at v.fire (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:61:1844)
    at a.fire (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:69:19042)
    at c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:111:17079
    at b.invoke (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:61:145)
    at p.deliver (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:61:2266)
    at v.fire (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:61:1844)
    at a.fire (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:69:19042)
    at MessagePortMain.<anonymous> (c:\Program Files\Microsoft VS Code Insiders\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:111:15119)
    at MessagePortMain.emit (node:events:526:28)
    at Object.MessagePortMain._internalPort.emit (node:electron/js2c/utility_process_init:5:364) circuitpython.library.update {"value":"joedevivo.vscode-circuitpython","_lower":"joedevivo.vscode-circuitpython"}
```

mirkoschubert, 2022-10-16:
**I maybe found a solution!** It seems that the LibraryManager tries to read all installed libraries and fails. So the `workspaceLibraries` won't be written at all and the error will occur.

So I tried two solutions to make this work again:

1. I had some custom libraries in the `lib` folder, so I moved them out - **no success**
2. Every file on my Adafruit board (in my case the Matrix Portal M4) had a companion file with `._`, e.g. `._code.py`. I deleted them. - **success**

I don't know what the `._` files are for and if they come either from this extension or from another, but deleting them helped to use the LibraryManager again.

tyeth, 2022-10-16:
Move the custom libraries back and see if it still works. I'm messing with libraries and have it symlinked with other build folders like ```.git .vscode .pio```. I don't think I have any underscore copies of files 🤔

mirkoschubert, 2022-10-16:
Yeah, I just restarted VS Code and the LibraryManager wouldn't work again...

I just experimented some more. It seems that the `._` files are generated when I copy libraries from the `adafruit_circlepython_bundle` manually. And moving all custom libraries out of the `lib` folder (without symlinking them back in) seems to do the trick. I could reproduce it by moving them back in.

But I don't see why the extension does that. Maybe it has to to with the `.mpy` format?

IrregularShed, 2022-12-06:
I've had this problem repeatedly, from the first time I tried the extension. I've got a project I'm working on that really lends itself to CircuitPython and I really don't like the alternative editor choices, so I've spent some time recovering from Covid (again) trying to fix this - or, at the very least, understanding it and finding a workaround.

I think I've succeeded in the second part. I'll need to get used to Typescript and extension authoring to do the first part (oh, and also the whole 'rona situation).

## The bug detection

Working backwards from the error alert, "Cannot read properties of undefined (reading 'name')", we can deduce that something is trying to work with `.name` on an object that hasn't been initialised correctly. So, I found the folder that the extension code resides in on my PC, and opened that up in VS Code. A global search for `.name` had a few matches in the Library Manager, which makes sense, so before each one I put a unique `console.log` line in. (I'm old-school, what can I say.) Back in the VS Code instance I've been using with CircuitPython I ran one of the commands that was failing for me and, when it did, opened up the Electron DevTools to see what would've triggered it.

The constructor for `LibraryQP` was where it was getting thrown, against the first parameter, so I searched for where `LibraryQP` was getting instantiated. This led me to `getInstalledChoices()` and `getUninstalledChoices()`, both of which weren't doing any sanity checking on the values it was getting from the list of libraries which appear to be coming from the file system. (Remember I mentioned covid? I got too worn out to investigate that further.) I added a check to these functions, and now it works.

## The proper fix

When I'm recovered, I'll do try and do this (unless some other hero wants a bash at it). I've been hacking around at the compiled javascript to make it work, and the proper fix needs to be done in the source typescript. I had a quick glance but need to be feeling better.

## The hacky fix

* Find the extension's installed directory. For me, that is `~/.vscode/extensions/joedevivo.vscode-circuitpython-0.1.19-win32-x64/out/librarymanager`.
* Open the file `out/librarymanager/libraryManager.js`
* Look for the `getInstalledChoices()` code. It's followed immediately by `getUninstalledChoices()`, which we'll also be modifying.
* Surround the line `choices.push(new LibraryQP(b, p));` with a conditional, like so:
```js
if (b) {
  choices.push(new LibraryQP(b, p));
}
```
* Alter the conditional in the next function to also check `b`, like so:
```js
if (b && !this.workspaceLibraries.has(v)) {
  choices.push(new LibraryQP(b, null));
}
```
* Save and close VS Code, then open your CircuitPython project - the library commands should now work.

Remember - this is only temporary. Any update to the extension will hose this directory, so hopefully the next upload will have fixes in it!

steka, 2023-01-17:
I have the same error, and I running CircuitPython 8.0.0 beta 6 on a Raspberry Pi Pico W.
Is it perhaps related to the patch I had to do to make Serial Monitor to work, see https://github.com/joedevivo/vscode-circuitpython/issues/103#issuecomment-1380991841

veggie2u, 2023-05-19:
Two separate issues. Well, one explanation, and one solved issue. The ._file does show up if a circuit python device is connected to my computer and  I manually move a file/folder on OSX to the circuit python device. This is a Mac thing. OSX is adding "extend file attributes" on a fat32 like device that can't accept the extended attributes OSX wants. So, not really and issue. Delete if you want.

The library issue is a bug. It is caused by using a file/library in the lib folder that is not in the Adafruit Circuit Python list of official libraries. In my case I was trying to use Bluepad32 which you can get from GitHub or the community download of libraries. When all the existing library files/folders are looped over, information is gleaned from the Adafruit repo about this library. However, like IrregularShed discovered, it is not found and is therefore undefined. I have a pull request that may or may not be merged that would fix this. It would display the library and version and indicate it is not undateable.

https://github.com/joedevivo/vscode-circuitpython/pull/122