joedevivo/vscode-circuitpython #103: [BUG] Cannot read properties of undefined (reading 'vid')

LazaroFilm, 2022-11-10:
**Describe the bug**
I have a RP2040-LCD-1.28 board with [custom compiled CircuitPython](https://github.com/fonix232/circuitpython/tree/feature/waveshare_rp2040_lcd_1_28/ports/raspberrypi/boards/waveshare_rp2040_lcd_1_28) installed. When trying to connect to the port, I get the error `Cannot read properties of undefined (reading 'vid')`

**Desktop (please complete the following information):**
 - MacOs Ventura 13.0
 - VsCode 1.73.1

LazaroFilm, 2022-12-16:
So after some digging, The board in question doesn't have an assigned valid VID/PID in its code yet. I have reached out to Waveshare and they said they were given an ID, but it has not been published yet. Until then, the plugin will not work. It's a shame we can't toggle off the VID/PID checkup from this plugin to use unidentified boards. 
https://github.com/adafruit/circuitpython/pull/7153

codejoeo, 2022-12-24:
I'm getting the same error, but my case is a bit different.  I'm using a Raspbery Pi Pico W. 
The serial monitor will connect when I'm running Circuitpython 7.3.3.  But when I upgrade to CircuitPython 8.0.0 Beta 6 I run into this same error.

gkovacsp, 2022-12-28:
I can second that it is not working with Pico W and CP 8.0.0 beta6
Although I'm pretty sure it worked for a while I just can't recall when it stopped.

byte4c, 2023-01-12:
I had the same issue with my Pico W but I've found a temporary workaround by manually adding it.

The following folder contains all the board-infos known by the extension
`C:\Users\<username>\.vscode\extensions\joedevivo.vscode-circuitpython-0.1.19-win32-x64\boards`

I copied the content of the Pico folder ( \0x239A\0x80F4) to the Pico W folder (\0x239A\0x8120).
I updated the board.pyi file to reflect it being a Pico W instead of a Pico by updating board name.

Secondly, in the boards folder is a 'metadata.json' file to wich I've added the Pico W data (based on Pico)
  ```
{
    "vid": "0x239A",
    "pid": "0x8120",
    "product": "Pico W",
    "manufacturer": "Raspberry Pi",
    "site_path": "raspberry_pi_pico_w",
    "description": "Raspberry Pi Pico W"
  },
```
Once this is done, I can select my COM port and it is detected as the Pico W and I'm able to make a serial connection.

This repo also contains files to generate these board info files and it gets it's information from the circuitpython repository. The repository in question already contains the required information for the Pico W so I'm hoping that with a new release of this extension the new/updated boards information will just be included.. In the meantime, you can add it manually.

I'm assuming these steps could also work for adding other boards. As long as you have the VID/PID info by which the USB device will identify itself, it should work.

LazaroFilm, 2023-01-12:
Thanks! Nice bandaid solution while we wait for a fix.

gkovacsp, 2023-01-13:
You can also use the following in a mac terminal:

`screen /dev/tty.usbmodem1301  115200`

Your `tty.usbmodem1301` might have a different name, just `ls /dev/tty*` to find it

djotaku, 2023-02-19:
Interestingly, I looked at @byte4c  solution, but when I do Search CircuitPython board - it's there. But might it be there without the vid parameters that the program is expecting?

LazaroFilm, 2023-02-19:
It has been added recently after I emailed Waveshare. I haven't tried it since they added the ID, I have somehow misplaced this board 😅

LazaroFilm, 2023-08-10:
This error means that the PID (Product ID) of your board is not registered in the [CircuitPython Product list](https://github.com/raspberrypi/usb-pid). You need to go there and ask for your board to be registered with a PID then you will be able to get it to work. Until then, you can use MicroPython or Arduino...

wyattbiker, 2023-08-10:
I am fine with using VSCode and tio as the terminal. Its just weird that the plugin requires this. As you can see in the settings of my screenshot, the PID and VID is available. Not sure why it has to be in another list.

todbot, 2023-08-15:
Also having this problem, on an officially supported CircuitPython board https://circuitpython.org/board/lolin_s3_mini/

Is there a way to turn off VID/PID detection and just have it open the dang serial port?

wyattbiker, 2023-08-15:
Either that or allow programmer to put it in some config file

LazaroFilm, 2023-08-17:
That issue got me out of CircuitPython. I'm now focusing on Cpp and MicroPython.

todbot, 2023-08-17:
> That issue got me out of CircuitPython. I'm now focusing on Cpp and MicroPython.

One can also turn just off the CircuitPython extension and use VSCode as a text editor for “code.py” and an open terminal window running “tio” to access the REPL. I find that works pretty well for my purposes.

wyattbiker, 2023-08-17:
You don't have to turn off the circuitpython plugin. Just make sure you name your **code.py** to **main.py** to avoid the naming conflict with the python stdlib version. **main.py** runs at boot same as **code.py** so no problem there. Then run **tio** for the REPL part from the vscode terminal. E.g. `tio /dev/ttyACM0`

tyeth, 2023-09-03:
To get the latest boards I forked the extension repo, and ran the scripts/build_stubs.sh which gave me an updated stubs and boards folder, copied over the extensions versions gives latest data. I've blocked those folders from gitignore and you can now just download a zip file of my fork, and copy the `stubs` and `boards` folders across.

I'm happy to update the fork whenever required, eventually with CI, but ping me if new board added to circuitpython and you want the latest boards/stubs.
https://github.com/tyeth/vscode-circuitpython/

CombatVolcano, 2024-07-26:
bug after bug after bug after problem
circuitpython in VS code does NOT WORK IN 2024