joedevivo/vscode-circuitpython #106: [BUG]  "h:\code.py" is overriding the stdlib module "code"

rpgrunwald, 2022-12-12:
**Describe the bug**
Error message "h:\code.py" is overriding the stdlib module "code"

**To Reproduce**
Steps to reproduce the behavior:
Add CIRCUTPY drive with Open Folder
Open code.py
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Version  10.0.19045 Build 19045
- Visual Studio Code 1.74.0
- vscode-circuitpython [0.1.19]

**Additional context**
Adafruit Circuit Playground Express with adafruit-circuitpython-circuitplayground_express-en_US-7.3.3.uf2

IrregularShed, 2022-12-21:
This issue is unrelated to the extension. It's occurring due to the way the Pylance language server is configured; Pylance has no concept of CircuitPython or MicroPython and is attempting to treat what it has found in the same way as it would 'proper' Python. I had the same experience, and fixed it like this:
* open Settings
* search for `Analysis: Diagnostic Severity Overrides` to get straight to the correct setting
* click `Edit in settings.json`, which should take you to the correct part of the `settings.json` file structure
* this is what mine says now:
```json
"python.analysis.diagnosticSeverityOverrides" : {
  "reportShadowedImports": "none"
}
```

Hope that helps!

joshmarinacci, 2023-02-12:
This might be a good thing to add to the extension docs. I ran into it as well.

stefan-sherwood, 2023-02-18:
According to the [documentation](https://learn.adafruit.com/welcome-to-circuitpython/creating-and-editing-code):

> CircuitPython looks for a code file on the board to run. There are four options: code.txt, code.py, main.txt and main.py. CircuitPython looks for those files, in that order, and then runs the first one it finds. While code.py is the recommended name for your code file, it is important to know that the other options exist.

So an arguably better solution is to name your main program file `main.py` instead of `code.py`. Then you'll still get (useful, IMO) warnings when you override modules but you won't have the problem reported here.

VashJuan, 2023-10-29:
Much better solution (IMHO) is just to **rename code.py to main.py**: it won't conflict with the standard libraries & will still get run on boot.