joedevivo/vscode-circuitpython #108: [BUG]

christofeugster, 2022-12-28:
**Describe the bug**
ModuleNotFoundError: No module named 'board'
In VSC Console and more important when trying to initialize VSC debugging
module 'board' is not found (i guess because it is on a different directory than VSC is looking for)

**To Reproduce**
Steps to reproduce the behavior:
1. Installed Circuitpyhon extension and (Python, Pylance, isort, Intellicode and API, Jupiter) 
2. Circuitpython and REPL works flawlessly except - Python Console has the ModuleNotFoundError
3. Start Debugging, creating new json launch file
4. Start Debugging
5. Error occurs

`Exception has occurred: ModuleNotFoundError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
No module named 'board'
  File "D:\main.py", line 3, in <module>
    import board
  File "C:\Users\****\AppData\Local\Programs\Python\Python311\Lib\runpy.py", line 88, in _run_code
    exec(code, run_globals)
  File "C:\Users\****\AppData\Local\Programs\Python\Python311\Lib\runpy.py", line 198, in _run_module_as_main (Current frame)
    return _run_code(code, main_globals, None,` 

**Expected behavior**
VSC (Python, Pylance) should search module 'board' in the extraPaths i guess.
c:\\Users\\****\\.vscode\\extensions\\joedevivo.vscode-circuitpython-0.1.19-win32-x64\\boards
but instead is looking in stock python folder or D:\ drive directly

**Desktop (please complete the following information):**
 - OS: Windows 11

**Additional context**
- Ive tried to include directories in Pylance Settings. No luck
- extra Paths are showing up in settings.json file. 
- In launch.json file i set "justMyCode": false - no difference

Any suggestions how to make VSC debugging work for Circuitpython?
Thanks and best regards
Chris

wmerkens, 2023-02-04:
If you go to the scripts directory and edit build-stubs.sh and at the bottom, rem out " rm -rf stubs/board" and run the script it will rebuild stubs/boards. then go replace it in .vscode/extensions/joedevivo.vscode-circuitpython-0.1.19-linux-x64/ the boards/stubs folders with the newly created one then this will go away.

Also if the date in the path line in settings.json does not match the current bundle you will get this problem, correct the date or erase .vscode and if have a workplace file erase it and let it all be rebuilt will correct it for a bad date scenario

cwgphillips, 2023-02-10:
> Use git and pull this repo and then go into the downloaded files, vscode-circuitpython/scripts
> 
> Then in the scripts directory go and edit build-stubs.sh and at the bottom, rem out " rm -rf stubs/board" and run the script it will rebuild stubs/boards. then go replace it in .vscode/extensions/joedevivo.vscode-circuitpython-0.1.19-linux-x64/ the boards/stubs folders with the newly created one then this will go away.
> 
> Also if the date in the path line in settings.json does not match the current bundle you will get this problem, correct the date or erase .vscode and if have a workplace file erase it and let it all be rebuilt will correct it for a bad date scenario

I found your YouTube vid as well as this - solved it for me, thanks @wmerkens !

KristenLeach, 2024-01-02:
Are there any plans to make this a permanent fix? I would much rather use vscode than mu but it's kind of counterintuitive if you have to edit the extension files manually.

wmerkens, 2024-01-02:
I would like to see a button in VSC that, when you click it, or an option you run via the F1 menu to rebuild stubs/boards and put them in place. Also, I don't know why the file "board" is removed.

CombatVolcano, 2024-07-26:
Same problem. I am new to this and I guess I'm assuming that you are not intended to be able to run circuitpython code within VS code or do actual debugging, which has me at a loss for words.

Is there really no fix for this besides the complicated stuff above? I didn't do anything weird I swear. This is normal install stuff.