joedevivo/vscode-circuitpython #114: [Feature Request] Mirror Directories

dmopalmer, 2023-01-14:
Editing a file directly on the target device is simple, but it is also error prone.  After you save it and unmount the device, you have no further access to the file until you plug the device in again.  You can't keep track of it in git, or put the same file on a second device.  If your drone controller flies over the horizon, it takes your code with it.

I suggest that it would be a good feature to allow files to be edited on the desktop hard drive, to be automatically copied over to the device whenever changed.  That way the files can reside in your local git-maintained directory.  Also, code can be written without requiring you to be tethered to the device.

Not all files on the desktop should be copied over.  (e.g. the .git directory, the .DS_Store files that MacOS litters everywhere, etc.)

One suggested implementation: A 'manifest' file listing the names of all files that should be mirrored to the device.  Possibly with the behavior of saying 'don't include the .git directory, don't include the manifest file, but everything else goes'.

Another possible implementation:  Everything in a subdirectory gets mirrored to the device with the same diskname.  (By default the device will be named CIRCUITPYTHON, but you can rename it, which is especially useful if you are developing a system of two devices that talk to each other and you want to edit files for both devices at once.)

0ge, 2023-01-31:
A workaround is to use the [Run on Save](https://marketplace.visualstudio.com/items?itemName=emeraldwalk.RunOnSave) extension and configure it to copy your files. Add this to the *workspace* settings:

```json
"emeraldwalk.runonsave": {
        "commands": [
            {
                "match": ".*\\.py",
                "isAsync": true,
                "cmd": "cp *.py /Volumes/CIRCUITPY/"
            }
        ]
    }
```

*(Note that if you are modifying lib-files or using subdirectories, you'll need to modify)*

Timmoth, 2023-03-13:
> A workaround is to use the [Run on Save](https://marketplace.visualstudio.com/items?itemName=emeraldwalk.RunOnSave) extension and configure it to copy your files. Add this to the _workspace_ [...]

[I just wrote a post for anyone looking for an easy way to do this on Windows
](https://timmoth.com/posts/kN22D0Z4okeIE8VMIQ9gwQ)

jkozniewski, 2024-01-04:
> A workaround is to use the [Run on Save](https://marketplace.visualstudio.com/items?itemName=emeraldwalk.RunOnSave) extension and configure it to copy your files. Add this to the _workspace_ [...]

@0ge - thanks for the tip, regarding your note - any idea how to configure it so it's gonna be iterative and include subdirectories / lib folder ?

0ge, 2024-01-06:
There are many options you could use. I haven't tested these, but…

- `cp -r . /Volumes/CIRCUITPY/` would copy the entire directory, including vscode settings, git directories, etc.
- `cp ${file} /Volumes/CIRCUITPY/` would just copy the file you just saved

Additionally, Run on Save is capable of running multiple commands, so you can combine these. See its documentation for more guidance.

And if you are on Windows, you might want to use [`xcopy`](https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/xcopy) instead.

axelmagnus, 2024-03-07:
also, some ideas in #144 that I closed now that I see that the request was already here