joedevivo/vscode-circuitpython #117: [BUG] Selecting Serial Port doesn't work

djotaku, 2023-02-19:
**Describe the bug**
When I try to select the serial port, I get teh error: cannot read properties of undefined (reading 'vid')

**To Reproduce**
Steps to reproduce the behavior:
1. CircuitPython:Select Serial Port
2. Click on /dev/ttyACM0
3. See error

**Expected behavior**
Connect to serial port

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Fedora 36 KDE Plasma 5
 - Version [e.g. 22] 0.1.20

**Additional context**
I know it's not a permissions error because I can use screen to connect to that serial port

Neradoc, 2023-02-19:
Looks like a duplicate of #103 ?

djotaku, 2023-02-19:
Yes, whoops. Sorry. Didn't do a good job of searching

closed 2023-02-19 by djotaku