joedevivo/vscode-circuitpython #118: Can't reopen serial monitor windows - on windows

freemansoft, 2023-02-23:
**Can't **
I can open a terminal to the serial port and interact with the repl the first time.  

**To Reproduce**
1. Open the Serial Console with ctrl-shift-p `CircuitPython open serial monitor`
2. Interact with device
3. Close the terminal session
4. Device stays connected to VS Code
5. There is no way to re-open the terminal session.
6. connecting and disconnecting in the bottom right corner does nothing

**Expected behavior**
Expect `CircuitPython open serial monitor` any time I select it from the menu.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version Latest VS Code

**Additional context**
I don't see a way to connect a terminal session to the connected COM port.

wuhao19920619, 2023-05-15:
I have the same issue. Also `Select Serial Port` in the lower right corner doesn't work.

With `Adafruit CircuitPython 8.1.0-beta.2-31-g827eaeb1f on 2023-05-12; VCC-GND YD-ESP32-S3 (N16R8) with ESP32S3` the Serial Monitor doesn't work at beginning even with `run as administrator`

patricksurry, 2024-01-21:
i see this on mac too.  closing and reopening the containing vscode window often seems to help

dorinclisu, 2026-01-18:
The problem is that there is no documented way to gracefully close the session, so when closing the terminal session, the serial session is probably orphaned.