joedevivo/vscode-circuitpython #134: [BUG] Updated to 0.2.0 from 0.1.20 cannot pick a board

wmerkens, 2024-01-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'update the extension'
2. Click on 'click bottom right "Choose Board"'
3. Scroll down to 'N/A'
4. See error

 command 'circuitpython.selectBoard' not found

**Expected behaviour**
Should open from the top of the VSCode, the board select list

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux KDE Neon
 - Version 22.04

**Additional context**

I wiped the .vscode from the circuitpy drive so as to rebuild it. I made sure the firmware was up to date.

"circuitpython.board.version": "8.2.9"

I tested with a second board, and I got the same error.

I reverted back to v0.1.20, and things work correctly.

DavidMKyle, 2024-01-01:
Seeing something similar here on W10 where was working at v0.1.20, now running v0.2.0 the VSC Extension Host Output shows:
```
2024-01-01 14:57:52.197 [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
2024-01-01 14:57:52.197 [error] Error: ENOENT: no such file or directory, open 'c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\boards\metadata.json'
```
If I supply that file using @wmerkens  method from #121 I get past this error and things load until the libs try and update, when this error is shown:
```
2024-01-01 15:05:49.605 [error] TypeError: The "path" argument must be of type string. Received null
	at new NodeError (node:internal/errors:399:5)
	at validateString (node:internal/validators:163:11)
	at Object.join (node:path:423:7)
	at LibraryManager.bundlePath (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\librarymanager\libraryManager.js:374:21)
	at LibraryManager.loadBundleMetadata (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\librarymanager\libraryManager.js:377:31)
	at Container.loadBundleMetadata (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\container.js:89:58)
	at LibraryManager.getBundle (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\librarymanager\libraryManager.js:365:31)
	at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
	at async LibraryManager.updateBundle (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\librarymanager\libraryManager.js:253:13)
	at async LibraryManager.initialize (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\librarymanager\libraryManager.js:106:9)
	at async Container.initialize (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\container.js:38:9)
	at async Container.newInstance (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\container.js:16:9)
	at async activate (c:\Users\David\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\extension.js:13:21)
	at async u.n (c:\Users\David\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:140:6255)
	at async u.m (c:\Users\David\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:140:6218)
	at async u.l (c:\Users\David\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:140:5675)
```

clickonben, 2024-01-03:
If you open the VSIX file as a zip file this file is not there `extension\boards\metadata.json` in V0.20 and so does not get installed to %USERPROFILE%\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\boards.

wmerkens, 2024-01-23:
Using VSC 1.85.2 and 1.2.0 now seems to work correctly, tested on three systems.