joedevivo/vscode-circuitpython #136: [BUG] Latest CircuitPython Extension (0.2.0) does not work with latest VSCode Version (1.85.1)

grosdode, 2024-01-05:
**Describe the bug**
If I try to execute a CircuitPython command in VSCode I get "Command 'xyz' not found" for all commands (open serial Monitor, update libraries, ...)

**To Reproduce**
Install the latest circuit CircuitPython Extension (0.2.0) and the latest VSCode (or VSCodium) Version (1.85.1) and try any command 
CircuitPython Extension (0.1.2) works fine with VSCode (or VSCodium) Version (1.85.1)
CircuitPython Extension (0.2.0) works fine with VSCode (or VSCodium) Version (1.84.2)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windowns 10 Pro

**Additional context**
Tested on 3 different Windows 10 pro computers

clickonben, 2024-01-06:
I think this is the same as issue #134.

If you open the VSCode dev tools you should see an error saying the CircuitPython extension cannot find `%USERPROFILE%\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\boards\metadata.json` which then causes the extension not to load and so all of the CircuitPython commands cause the error in the screenshot.

If you inspect the contents of the .vsix file (it's just a zip file), you will see that this file is missing from `extension\boards\` as compared with the previous version .vsix file.

You can get past this and get the extension to load by placing the `metadata.json` file from the previous version in the folder `%USERPROFILE%\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\boards\`

However, when I did this there were further bugs related to loading libraries as outlined in issue #134

grosdode, 2024-01-06:
Yes I get the same error:
Activating extension 'joedevivo.vscode-circuitpython' failed: ENOENT: no such file or directory, open 'c:\Users\xxx\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\boards\metadata.json'.
[image]
But I'm wondering why it worked with an older VSCode version

clickonben, 2024-01-06:
> But I'm wondering why it worked with an older VSCode version

How sure are you that it did?  

VS Code 1.85.1 came out before CircuitPython Extension (0.2.0), so to have CircuitPython Extension 0.2.0 installed in VSCode 1.84.2 would mean that they were updated out of order.

As far as I can tell CircuitPython Extension (0.2.0) is just broken but if you figure out how it could have worked in VSCode 1.84.2, it would be interesting to know.

grosdode, 2024-01-06:
Pretty sure. 
I had VSCode and VSCodium installed. VSCode was up to date but VSCodium still on 1.84.2 (do not use it verry often) and I had to download and install the update by hand. Before the update everything works fine with 0.2.0 of CircuitPython and VSCodium and I install extension by extension from my VSCode to VSCodium, to check if any of the extensions causes this problem. After I installed all possible extensions it worked still fine. Than I saw that VSCodium was not up to date. So I did the update by hand and CircuitPython 0.2.0 no longer works.

dzyla, 2024-01-08:
Same here, latest version of vscode.

Rambomst, 2024-01-10:
> I think this is the same as issue #134.
>
> If you open the VSCode dev tools you should see an error saying the CircuitPython extension cannot find [...]

This fixed part of my issue but I also didn't have python installed, once it was installed and I restarted this issue went away.

ScottMonaghan, 2024-01-15:
I had this issue and had to back install to 0.1.20:
[image]

Once I did that everything worked as expected.

toddmo, 2024-01-23:
uninstalled it. I never could get this to work regardless of version. latest VS Code on Linux Mint.

steka, 2024-01-26:
I had the same issue, and had to revert to v0.1.20 to get it to work again.

wmerkens, 2024-01-26:
Currently, on three Linux systems, I have 1.85.2 and 0.2.0 are working.

wiredopposite, 2024-02-02:
> uninstalled it. I never could get this to work regardless of version. latest VS Code on Linux Mint.

I had to install 0.1.2, then go into .vscode/extensions and manual delete 0.2 for it to work

t-makaro, 2024-03-19:
Just encountered this issue on Windows 11. Downgrading to v0.1.20 worked.

Version: 1.87.2 (user setup)
Commit: 863d2581ecda6849923a2118d93a088b0745d9d6
Date: 2024-03-08T15:20:17.278Z
Electron: 27.3.2
ElectronBuildId: 26836302
Chromium: 118.0.5993.159
Node.js: 18.17.1
V8: 11.8.172.18-electron.0
OS: Windows_NT x64 10.0.22631

karunt, 2024-06-01:
Has this been fixed yet for version 0.2.0?

wmerkens, 2024-06-01:
I am afraid nope, not fixed. Also, I have found the script used to rebuild the stubs/boards needs to be reworked since it does not build stubs correctly anymore. I am looking at the code of the extension, but I have never done typescript, so at the moment, I am learning how the extensionis put together and built.

RekoSK, 2024-09-20:
> I think this is the same as issue #134.
>
> If you open the VSCode dev tools you should see an error saying the CircuitPython extension cannot find [...]

So you're saying it's just better to stick with the old versions?

ajhendrix25, 2025-10-23:
> I had this issue and had to back install to 0.1.20: 
[image]
> 
> Once I did that everything worked as expected.

Anyone seeing this as of Oct 23, 2025, its still the solution

wmerkens, 2025-10-23:
I have forked this repo and fixed a lot of issues, see https://github.com/wmerkens/vscode-circuitpython

ajhendrix25, 2025-10-23:
@wmerkens I did end up finding your version upon further investigation. I pulled the circuitpython docs and I guess I must have skipped the first page where it specified to install your version. Thank you for forking it and for a timely response here.