joedevivo/vscode-circuitpython #139: [BUG] command 'circuitpython.library.fetch' not found

Quinnsicle, 2024-02-02:
**Describe the bug**
The extension does not initialize properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Open workspace
2. use command palette and run 'CircuitPython: Check for latest bundle'
3. See error

**Expected behavior**
It appears I'm somehow missing a metadata.json in the boards directory

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 10.0.22631 Build 22631

**Additional context**
I found the relevant error log: 
`Error: ENOENT: no such file or directory, open 'c:\Users\Quinn\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\boards\metadata.json'`

Full Extension Host Output:
```
2024-02-02 17:05:04.970 [info] Extension host with pid 40736 started
2024-02-02 17:05:04.971 [info] Skipping acquiring lock for c:\Users\Quinn\AppData\Roaming\Code\User\workspaceStorage\ff591bdbcb561092c543c8b9ad7b87ab.
2024-02-02 17:05:05.031 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2024-02-02 17:05:05.034 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2024-02-02 17:05:05.045 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2024-02-02 17:05:05.232 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2024-02-02 17:05:05.233 [info] ExtensionService#_doActivateExtension ms-python.python, startup: true, activationEvent: 'workspaceContains:/code.py', root cause: joedevivo.vscode-circuitpython
2024-02-02 17:05:05.280 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2024-02-02 17:05:05.293 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2024-02-02 17:05:05.386 [info] ExtensionService#_doActivateExtension joedevivo.vscode-circuitpython, startup: true, activationEvent: 'workspaceContains:/code.py'
2024-02-02 17:05:05.630 [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
2024-02-02 17:05:05.630 [error] Error: ENOENT: no such file or directory, open 'c:\Users\Quinn\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\boards\metadata.json'
	at Object.openSync (node:fs:605:3)
	at Object.func [as openSync] (node:electron/js2c/asar_bundle:2:2131)
	at Object.readFileSync (node:fs:473:35)
	at Object.readFileSync (node:electron/js2c/asar_bundle:2:9462)
	at Board.loadBoards (c:\Users\Quinn\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\boards\board.js:24:27)
	at new Container (c:\Users\Quinn\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\container.js:27:23)
	at Container.newInstance (c:\Users\Quinn\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\container.js:14:35)
	at activate (c:\Users\Quinn\.vscode\extensions\joedevivo.vscode-circuitpython-0.2.0-win32-x64\out\extension.js:13:49)
	at z.ib (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:156:13232)
	at z.hb (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:156:12946)
	at c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:156:11081
	at async c.n (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:140:6256)
	at async c.m (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:140:6219)
	at async c.l (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:140:5676)
2024-02-02 17:05:05.630 [info] Eager extensions activated
2024-02-02 17:05:05.640 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2024-02-02 17:05:05.642 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2024-02-02 17:05:05.646 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'api', root cause: ms-python.python
```

Quinnsicle, 2024-02-03:
I rolled back to the previous version, it had the metadata.json in the directory mentioned in the error message. I copied the metadata.json to the newer version's (2.0) boards folder and that error message went away.

---
**I got another error message:**
`2024-02-02 21:10:19.508 [error] Activating extension joedevivo.vscode-circuitpython failed due to an error:
2024-02-02 21:10:19.509 [error] Error: ENOENT: no such file or directory, open 'c:\Users\Quinn\AppData\Roaming\Code\User\globalStorage\joedevivo.vscode-circuitpython\bundle\20240130\adafruit-circuitpython-bundle-20240130.json'`
I resolved this by deleting the bundle folder (c:\Users\Quinn\AppData\Roaming\Code\User\globalStorage\joedevivo.vscode-circuitpython\bundle\20240130) and reloading vs code. The new bundle was downloaded again and the error message went away.

---
**I got another message after that...**
`2024-02-02 21:11:09.307 [error] TypeError: The "path" argument must be of type string. Received null`
I resolved this one by following @cshields [advice](https://github.com/joedevivo/vscode-circuitpython/issues/46#issuecomment-835392146) and added my board path (c:\\Users\\Quinn\\.vscode\\extensions\\joedevivo.vscode-circuitpython-0.2.0-win32-x64\\boards\\0x239A\\0x80CA) to the settings.json

---
Everything is working for me know. Hope this helps.

rishabhrkaushik, 2024-02-04:
I had the same issue. Following steps above by @Quinnsicle resolved it although it doesn't list my board in board selection. Looks like the older metadata (updated 11 months back) didn't have my board listed.