joedevivo/vscode-circuitpython #141: [BUG] no comands are found

hunter72007, 2024-02-11:
Navigate to your workspace.
Press Ctrl+Shift+P.
Attempt to run any CircuitPython command.
Observe the error.
Expected behavior
When attempting to run any command, I expect to receive a "command not found" error.

Screenshots
Screenshot

OS: Windows 11

shanselman, 2024-02-18:
Seeing this also @joedevivo

arthurolivierfortin, 2024-03-01:
got the same problem..

dustinlacewell, 2024-03-17:
This seems pretty blocking.,,

HarisM26, 2024-03-23:
had this problem today

scottpletcher, 2024-04-01:
Just started occurring for me this evening after, strangely, a week of flawless developing.  The only thing that I did was flash my Feather ESP32-S2 to CircuitPython 9.0.0.  That seemed to be the activity that started this.  Even tried to revert to 8.x..now not really sure what might have happened with either VSCode or my Feather.   Running VSCode 1.87.2 on Fedora 39 Silverblue (RPM install and not the Flatpak).

scottpletcher, 2024-04-01:
I seemed to have got rid of this issue by reverting back to v0.1.2, restarting VSCode and then upgrading again to v0.2.0.  🤞

Sola85, 2024-04-15:
Also facing this issue. Switching back to v0.1.20 seems to fix the issues for now.

cybermanhao, 2024-06-17:
same question,in windows11,i have use this extention correctly for 2days,but now this occured.
tring to install v0.1.2 and update solved it and then crashed again.
this time install v0.1.2 and update didnt solve it.