joedevivo/vscode-circuitpython #151: [BUG] Circuitpython basically just doesn't work in VS code anymore in 2024

CombatVolcano, 2024-07-26:
**Describe the bug**
Try following basic install on windows from the ground up. 
Install the latest python, latest VS code, the latest circuitpython extension.
Select python interpreter and restart VS code.
Try running select serial port now from the quick menu. It fails to find the file to execute the command.

There is no reason this should be so obtuse.

THIS JUST PLAIN DOES NOT WORK!

edwin721, 2024-08-07:
That seems to work with the previous version...

NicholasCessna, 2024-09-13:
This is still going on. "command not found" is there a resolution for this?

adamneighbors, 2024-09-14:
@notbrain @NicholasCessna Are y'all actually opening your VScode workspace on the board? I had the same problems until I opened my workspace on the actual board.

NicholasCessna, 2024-09-14:
@adamneighbors It says it when I am trying to find the serial connection to access the board. I can't get to the board via circuitpython.

RekoSK, 2024-09-20:
@adamneighbors , I just tried what you said, and it's not working. Progress is that i get a pop up saying activating extensions, and circuit python is "opening", but probably then something fails, because when i try to open the serial monitor for example, it still don't work (having the same error as everybody has). Did anybody found a solve for this?

CombatVolcano, 2024-09-20:
@RekoSK From the looks of it, this is the end of the road for VS Code and CircuitPython. AdaFruit who drive most support for CircuitPython these days want you to use Mu (I know). Everything is being deprecated and sequestered and with the insane selfishness of modern open source licensing that community is dying, too.

RekoSK, 2024-09-21:
@CombatVolcano I fixed it with only possible solution, and that is downgrading to 1.2.0. Right now everything works for me, but I only tried printing out hello word on OLED display.

Tbh, mu sucks (or I just don't know how to use it properly). When I add any library to my Raspberry pi pico, it doesn't automatically start autocorrecting and suggestions don't pop up. I tried searching the web, but none of the answers worked. 

On vs code however, it starts automatically autocorrecting me after I upload the library to the lib folder on the Raspberry. Idk how that works or anything, but even if it stopped working I can normally just install it through pip (which is another thing idk how to do on MU).

Ye and I LOVE the way vs 2019 and vs code is highlighting code, so thats another reason why I don't like using anything else 😄.

Duckferd, 2024-09-26:
I don't understand the fatalistic/nihilistic viewpoints about the future of Circuitpython, all I am looking for is a resolution so I don't have to continue using Mu (please and thank you to the author and community!). Both Micro and Circuitpython offer specific libraries that are both very useful and I would like to continue using both.

wmerkens, 2024-09-28:
This can be fixed in version 0.2.0; it just needs its boards and stubs rebuilt. See issue #148, where I describe a change to the build script that will allow you to rebuild the stubs and boards.

Basic steps: fix the script, run it and build the new stubs & boards, copy the two folders to the vscode/extention/joe... directory, and restart vscode.

I have 1.93.1 code and 0.2.0 extension working just fine.